gaoxiang12/slambook #4: about chapter 3

x007dwd, 2016-11-08:
I have tested the time consumption of the qr composition and direct inverse, it is not performed as the author commented(the qr compostion is more fast). my test code is here.
https://github.com/x007dwd/slambook/blob/opencv/ch3/useEigen/eigenMatrix.cpp
the matrix size is changed form 3 to 100

gaoxiang12, 2016-11-08:
I have tested your code, here is my result:
Eigen values = -0.827453
 0.533448
  1.35323
Eigen vectors =  -0.299578  -0.762914  -0.572901
  0.294233  -0.645089   0.705186
  0.907568 -0.0426913  -0.417729
time use in normal inverse is 182.134ms
time use in Qr compsition is 7.203ms

So it suggests QR is faster than direct inverse, same as Eigen suggests. Please check this page:  http://eigen.tuxfamily.org/dox/group__TutorialLinearAlgebra.html. How about your results? 

Since we are using fixed size matrix, perhaps Eigen has set the max size of it. If you want larger matrix you may use the dynamic size matrix.

x007dwd, 2016-11-08:
My result is here
Eigen values = 
-0.827453
0.533448
1.35323
Eigen vectors = 
 -0.299578  -0.762914  -0.572901
  0.294233  -0.645089   0.705186
  0.907568 -0.0426913  -0.417729
time use in normal invers is 11.543ms
time use in Qr compsition is 23.428ms

gaoxiang12, 2016-11-08:
I changed the Debug version into Release, please check it again. Do you use a virtual machine or a full ubuntu OS?

x007dwd, 2016-11-08:
In release code, it is right now, Thank you very much, I use a full Ubuntu OS instead of a virtual machine.

time use in normal inverse is 1.984ms
time use in Qr composition is 0.669ms

closed 2016-11-11 by gaoxiang12