gaoxiang12/slambook #6: ch6 g2o_curve_fitting 运行报错

AlwaysLearningJary, 2017-03-22:
第六章的g2o_curve_fitting在运行时报错，前面生成x,y点是正确的，到图优化的部分显示：
bool g2o::OptimizableGraph::Edge::resolveParameters(): edge not registered with a graph
addEdge: FATAL, cannot resolve parameters for edge 0x17a719
solve time cost = 2.103e-05 seconds. 
estimated model: 0 0 0

配置环境：ubuntu14.14, github上下载的最新的g2o,最新的eigen3,以及g2o需要的qt5.

gaoxiang12, 2017-03-23:
请检查代码输入是否正确，可能忘记添加了addEdge？

AlwaysLearningJary, 2017-03-24:
全部加了，而且程序是直接download github上的源代码编译执行的。不知道什么原因。
而且我在运行ch7的pose_estimation_3d2d时程序就崩溃，我打印了日志发现是在运行：
“optimizer.addParameter ( camera );”这句代码时程序崩溃，显示错误是：
*** Error in `./pose_estimation_3d2d': malloc(): memory corruption: 0x000000000266a430 ***
Aborted (core dumped)

也是重新download 源代码编译运行的。。。其他程序都能正常运行，就是跟g2o相关的程序会出错。看了一圈也不知道怎么弄。。。是最新的g2o不支持原来的addedge，addparameter方法了吗？

closed 2017-05-22 by gaoxiang12