gaoxiang12/slambook #10: project 运行报错

Allen-jl, 2017-04-07:
高博您好，我在实验第九章/0.2时，按照您的要求编译好了OpenCV3， 在数据集中生成了associate.py 文件，但运行bin/run_vo config/default.yaml 显示：segmentation fault
请问这可能是哪里出现了问题？
谢谢

gaoxiang12, 2017-05-22:
未确认，请在程序中放置断点以检查问题出在哪里。

simple1082, 2017-06-22:
楼主解决问题了吗，我的也是出现段错误
Using host libthread_db library "/lib/i386-linux-gnu/libthread_db.so.1".
Core was generated by `bin/run_vo config/default.yaml'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0xb77a4501 in myslam::Frame::Frame(long, double, Sophus::SE3, std::shared_ptr<myslam::Camera>, cv::Mat, cv::Mat) () from /home/dan/work/project/0.4/lib/libmyslam.so
(gdb) bt
#0  0xb77a4501 in myslam::Frame::Frame(long, double, Sophus::SE3, std::shared_ptr<myslam::Camera>, cv::Mat, cv::Mat) () from /home/dan/work/project/0.4/lib/libmyslam.so
#1  0xb77a47ef in myslam::Frame::createFrame() () from /home/dan/work/project/0.4/lib/libmyslam.so
#2  0x0804b4d6 in main ()

lbwlele, 2017-08-19:
请检查ubuntu版本，要求为ubuntu‘14.04，而不能是ubuntu kylin 14.04

xueliang2007, 2017-10-31:
我也出现了这个问题，我是16.04的，仿真的时候报错如下
The inferior stopped because it received a signal from the Operating System
 Signal name : SIGSEGV
 Signal meaning : Segmentation fault，目前我是无解了

vampire2017, 2017-11-01:
应该是你的程序哪边弄错了,,16.04亲测,,已过

jackxujh, 2019-04-09:
#27

youngsilence, 2019-11-05:
> 在 #27 和 #76 分别有不同的解决方法，适合不同的情况。

测试过了好几个方法，都没成功