gaoxiang12/slambook #12: Book Language

JustWon, 2017-04-20:
Hello
I am pleased to hear that the book will be released this month.
BTW, does that book is written in English or Chinese?
I am one of the people who is looking forward your book.
Thank you.

gaoxiang12, 2017-04-21:
Hi, we currently have Chinese version only, but as a foreigner friend is helping me to translate it into English, it is hopeful to have a English version in some months. 
Thank you for your support.

xeechou, 2017-05-08:
Will it available as ebook?

gaoxiang12, 2017-05-08:
@xeechou 
Yes, see this page: https://gaoxiang12.github.io/slambook-en/ 
Feel free to contribute if you have time, as I am not a native English speaker.

JustWon, 2018-09-18:
Hello again.
I am the manager of [SLAM Research KR](https://www.facebook.com/groups/slamkr/) group in Facebook. 
I bought your Chinese SLAM book from [here](https://www.chinahao.com/Product/549114804702/!_visual_slam_fourteen:_from_theory_to_practice__gao_xiang_gao_bo__and_other_slam_introductory_tutorials__slam_basic_tutorials__slam_programming_tutorial_algorithm_visual_machine_learning).

I was impressed by the contents of the book.
It's very informative and beneficial for the novice of SLAM research.
Even though I cannot read Chinese, I can somewhat understand via Google translator.

Here is my question, can I share the Google translated version(Korean) of this book?
(which means not perfect, but I would revise and correct the weird sentences from Google translator)
You are the author of this book, so I might need to get an acceptance from you.
I guess you are planning to distribute the English version of this book,
I want to propagate the knowledge to the Koreans since this book is awesome.
However, if you don't accept it, I will not.

Thanks.

gaoxiang12, 2018-09-20:
Hi @JustWon , glad to know you like my book. Please feel free to do the translation since i don't have any official publication plan for an English version. Actually I'm doing the second version of this book which may provide more detailed experiments and mathematics derivations. After that I'll spend some time on English translation, and I also need to translate the figures and code comments.

I may also provide the latex source code for you if you need it, which may save some time for you.

JustWon, 2018-09-20:
@gaoxiang12  
Thank you very much.
I am so glad to hear that.
If you can provide the latex source of the book, It would be a great help to me and our community.
dongwonshin@gist.ac.kr
This is my email address.
Again, Thank for your generous support.
Sincerely

gaoxiang12, 2018-09-24:
@JustWon I've already sent the email to you, please check if you have received it.

JustWon, 2018-09-26:
@gaoxiang12 
Thank you.
I received your email.
I would continuously translate the book into Korean.
Again, thank you for your cooperation.

JustWon, 2019-02-23:
Hello,
I completed the translation in Korean and I am planing to update it to the github repo.
https://github.com/slam-research-group-kr/slambook-ko
I updated some parts of the book according to your slambook-en template.
but I suddenly came up with the idea that "can I use the pictures from your book?".
plz check the repository and give me your comments.
Thank you!

gaoxiang12, 2019-02-25:
Hi Dong-Won Shin,
It's ok to put the figures on your github. Do you need to translate the
figures or just leave them in Chinese? Let me know if you need any help
from my side.

Cheers,
Xiang

Dr. Xiang Gao (高翔)
Ph. D of Tsinghua University
Postdoc of Technical University of Munich
Autonomous Driving Unit, Baidu, Beijing, China
Tel: (+86) 18811360537