gaoxiang12/slambook #13: Conversion from pangolin::OpenGlMatrix to Eigen::Matrix

Jyang772, 2017-05-07:
I am getting the following error when trying to build visualizeGeometry:
```
error: conversion from ‘pangolin::OpenGlMatrix’ to non-scalar type ‘Eigen::Matrix<double, 4, 4>’ requested
         Matrix<double,4,4> m = matrix;
```

Jyang772, 2017-05-07:
Fixed.

`Matrix<double,4,4> m = ToEigen<double>(matrix);`

closed 2017-05-07 by Jyang772

jamboo2017, 2017-09-12:
It shows 
 error: ‘ToEigen’ was not declared in this scope
Any header files needed to use ToEigen()?
Thx

hoastyle, 2018-03-21:
Add #define HAVE_EIGEN and pangolin::ToEigen to fix this.

GoldExperience, 2018-11-18:
it gives error: ‘ToEigen’ is not a member of ‘pangolin’
when I use : Matrix<double,4,4> m =pangolin::ToEigen<double>(matrix);
and the same porblem as jamboo2017

missdown, 2019-02-15:
The latest version has fix the bug.

Lyoid, 2019-03-11:
you should install Eigen first.and then install pangolin.This way will help the pangolin add the function about Eigen.

wheatjo, 2019-09-23:
use this:
        for(int i = 0; i<4; i++)
            for(int j = 0; j<4; j++)
                m(i,j) = matrix(i,j);

wangman123, 2021-08-07:
Define the toeigen function yourself

xuehu2, 2023-09-26:
> you should install Eigen first.and then install pangolin.This way will help the pangolin add the function about Eigen.

this is a true way to solve the question.