gaoxiang12/slambook #14: Sophus库无法正常编译安装，导致凡是有Sophus库的项目都不能正常运行？

menglitanhua, 2017-05-11:
CMake Warning (dev) at CMakeLists.txt:76 (GET_TARGET_PROPERTY):
  **Policy CMP0026 is not set**: Disallow use of the LOCATION target property.
  Run "cmake --help-policy CMP0026" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The LOCATION property should not be read from target "Sophus".  Use the
  target name directly with add_custom_command, or use the generator
  expression $<TARGET_FILE>, as appropriate.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at CMakeLists.txt:57 (ADD_LIBRARY):
  Policy CMP0038 is not set: Targets may not link directly to themselves.
  Run "cmake --help-policy CMP0038" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  Target "Sophus" links to itself.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
CMake Warning (dev) at CMakeLists.txt:57 (ADD_LIBRARY):
  Policy CMP0038 is not set: Targets may not link directly to themselves.
  Run "cmake --help-policy CMP0038" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  Target "Sophus" links to itself.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at CMakeLists.txt:57 (ADD_LIBRARY):
  Policy CMP0038 is not set: Targets may not link directly to themselves.
  Run "cmake --help-policy CMP0038" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  Target "Sophus" links to itself.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at CMakeLists.txt:57 (ADD_LIBRARY):
  Policy CMP0038 is not set: Targets may not link directly to themselves.
  Run "cmake --help-policy CMP0038" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  Target "Sophus" links to itself.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /home/gr/Sophus/build
****
因为是新人不知道如何修改Sophus库使其正常运行，求教大神。。

gaoxiang12, 2017-05-11:
这段并没有任何错误，请直接make即可。

menglitanhua, 2017-05-11:
恩，我直接make 和安装之后，测试ch4的那个Sophus库事例，也能正常编译，但在生成可执行文件的时候出现错误。提示很多SO3/SE3相关的变量函数都没有定义，不知道是什么原因？猜测是不是因为上述安装Sophus库出现的那种警告，而使CMakeLists.txt的
# 为使用 sophus，您需要使用find_package命令找到它
find_package( Sophus REQUIRED )
include_directories( ${Sophus_INCLUDE_DIRS} )
这两句话不能正确定位其定义文件而造成的？说实话第一次跑这个例子的时候是正常的，自从我注意到那个warning之后，所有有Sophus库参与的例子都会出现这种错误。希望高博士帮帮忙，我已在这个问题上纠结了一个星期了。。。。

gr@daguoge:~/文档/slam文件/slambook-master/ch4/useSophus/build$ make
Scanning dependencies of target useSophus
[ 50%] Building CXX object CMakeFiles/useSophus.dir/useSophus.cpp.o
[100%] Linking CXX executable useSophus
CMakeFiles/useSophus.dir/useSophus.cpp.o：在函数‘main’中：
**useSophus.cpp:(.text+0xd1)：对‘Sophus::SO3::SO3(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&)’未定义**的引用
**useSophus.cpp:(.text+0xf4)：对‘Sophus::SO3::SO3(double, double, double)’未定义的引用
useSophus.cpp:(.text+0x126)：对‘Sophus::SO3::SO3(Eigen::Quaternion<double, 0> const&)’未定义的引用
**useSophus.cpp:(.text+0x1d2)：对‘Sophus::SO3::log() const’未定义的引用
useSophus.cpp:(.text+0x232)：对‘Sophus::SO3::hat(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用**
**useSophus.cpp:(.text+0x27c)：对‘Sophus::SO3::hat(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x295)：对‘Sophus::SO3::vee(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&)’未定义的引用
useSophus.cpp:(.text+0x341)：对‘Sophus::SO3::exp(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用**
useSophus.cpp:(.text+0x361)：对‘Sophus::SO3::operator*(Sophus::SO3 const&) const’未定义的引用
useSophus.cpp:(.text+0x41a)：对‘Sophus::SE3::SE3(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x43a)：对‘Sophus::SE3::SE3(Eigen::Quaternion<double, 0> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x4cf)：对‘Sophus::SE3::log() const’未定义的引用
useSophus.cpp:(.text+0x52f)：对‘Sophus::SE3::hat(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x586)：对‘Sophus::SE3::hat(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x59f)：对‘Sophus::SE3::vee(Eigen::Matrix<double, 4, 4, 0, 4, 4> const&)’未定义的引用
useSophus.cpp:(.text+0x642)：对‘Sophus::SE3::exp(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x662)：对‘Sophus::SE3::operator*(Sophus::SE3 const&) const’未定义的引用
useSophus.cpp:(.text+0x67b)：对‘Sophus::SE3::matrix() const’未定义的引用**
CMakeFiles/useSophus.dir/useSophus.cpp.o：在函数‘Sophus::operator<<(std::ostream&, Sophus::SO3 const&)’中：
useSophus.cpp:(.text._ZN6SophuslsERSoRKNS_3SO3E[_ZN6SophuslsERSoRKNS_3SO3E]+0x2e)：对‘Sophus::SO3::log() const’未定义的引用
collect2: error: ld returned 1 exit status
CMakeFiles/useSophus.dir/build.make:94: recipe for target 'useSophus' failed
make[2]: *** [useSophus] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/useSophus.dir/all' failed
make[1]: *** [CMakeFiles/useSophus.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

gaoxiang12, 2017-05-11:
请检查：(1)是否使用了模板类Sophus (2)是否在其他地方使用了Sophus (3)是否正确链接了Sophus库文件。你甚至可以手动指定头文件目录和库文件名称。

menglitanhua, 2017-05-11:
谢谢你的提醒，我弄好了。不过，我发现了个奇怪现象，忽略Sophus编译安装时的警告信息，在虚拟机下的Ubuntu16.04中，你的ch4程序可以正常运行，不需要做任何修改；而如果在物理机上的同样操作系统下，需要做如下两处修改才能运行。我怀疑还是因为Sophus库编译安装时违背那两处政策，如CMP0026，造成了find_package和${Sophus_INCLUDE_DIRS}不能正常获取Sophus的动态链接库路径而引起的。鄙人既是SLAM新手，同时也是Linux的新手，猜测而已。再次感谢你的及时回复，跪谢，跪谢。

cmake_minimum_required( VERSION 2.8 )
project( useSophus )

# 为使gen头文件
**include_directories( "/usr/include/eigen3" )**
#用 sophus，您需要使用find_package命令找到它
#find_package( Sophus REQUIRED )
#include_directories( ${Sophus_INCLUDE_DIRS} )

add_executable( useSophus useSophus.cpp )
**target_link_libraries( useSophus "/usr/local/lib/libSophus.so" )**

gaoxiang12, 2017-05-12:
非模板的sophus在编译时候并不需要安装，会自动找到所在的目录。我在1604下写程序时没有碰到过你这个错误。
事实上你可以检查find_package输出的Sophus头文件和库文件的变量值是否正确，来判断find_package是否正确调用。
问题关闭，祝好

closed 2017-05-12 by gaoxiang12

roachsinai, 2018-11-29:
> 非模板的sophus在编译时候并不需要安装，会自动找到所在的目录。

不知能够告知您的`Sophus`存放位置与`ch4`存放位置的关系？