gaoxiang12/slambook #18: ch5 CMakeList.txt error

actionLUO, 2017-05-19:
高博您好：请教一个问题~~~
我使用ubuntu16.04，build时显示：
/usr/bin/ld: cannot find -lvtkproj4
collect2: error: ld returned 1 exit status
CMakeFiles/joinMap.dir/build.make:350: recipe for target 'joinMap' failed
make[2]: *** [joinMap] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/joinMap.dir/all' failed
Makefile:83: recipe for target 'all' failed
make[1]: *** [CMakeFiles/joinMap.dir/all] Error 2
make: *** [all] Error 2
*** Failure: Exit code 2 ***
怎么解决？

gaoxiang12, 2017-05-19:
这是Ubuntu16.04下PCL的一个Bug，见：[https://haoyu.love/blog371.html] 和 [https://bugs.launchpad.net/ubuntu/+source/vtk6/+bug/1573234]
总而言之，你需要在CMakeLists.txt中添加一句：
list(REMOVE_ITEM PCL_LIBRARIES "vtkproj4")
才能使得编译通过。

gaoxiang12, 2017-05-21:
已在github issue中回复。

祝好

Xiang Gao (高翔)
Ph. D. Student in Tsinghua University
Room 428, Main Building, Beijing, 100084
Tel: (+86)18811360537