gaoxiang12/slambook #20: ch4 use Sophus 编译错误

robotdream, 2017-05-23:
ubuntu 16.04, 用的是非模板版本Sophus库，错误如下，请帮忙看一下，谢谢！

[ 50%] Linking CXX executable useSophus
CMakeFiles/useSophus.dir/useSophus.cpp.o: In function `main':
useSophus.cpp:(.text+0xd1): undefined reference to `Sophus::SO3::SO3(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&)'
useSophus.cpp:(.text+0xf4): undefined reference to `Sophus::SO3::SO3(double, double, double)'
useSophus.cpp:(.text+0x126): undefined reference to `Sophus::SO3::SO3(Eigen::Quaternion<double, 0> const&)'
useSophus.cpp:(.text+0x1d2): undefined reference to `Sophus::SO3::log() const'
useSophus.cpp:(.text+0x232): undefined reference to `Sophus::SO3::hat(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)'
useSophus.cpp:(.text+0x27c): undefined reference to `Sophus::SO3::hat(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)'
useSophus.cpp:(.text+0x295): undefined reference to `Sophus::SO3::vee(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&)'
useSophus.cpp:(.text+0x341): undefined reference to `Sophus::SO3::exp(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)'
useSophus.cpp:(.text+0x361): undefined reference to `Sophus::SO3::operator*(Sophus::SO3 const&) const'
useSophus.cpp:(.text+0x41a): undefined reference to `Sophus::SE3::SE3(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)'
useSophus.cpp:(.text+0x43a): undefined reference to `Sophus::SE3::SE3(Eigen::Quaternion<double, 0> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)'
useSophus.cpp:(.text+0x4cf): undefined reference to `Sophus::SE3::log() const'
useSophus.cpp:(.text+0x52f): undefined reference to `Sophus::SE3::hat(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)'
useSophus.cpp:(.text+0x586): undefined reference to `Sophus::SE3::hat(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)'
useSophus.cpp:(.text+0x59f): undefined reference to `Sophus::SE3::vee(Eigen::Matrix<double, 4, 4, 0, 4, 4> const&)'
useSophus.cpp:(.text+0x642): undefined reference to `Sophus::SE3::exp(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)'
useSophus.cpp:(.text+0x662): undefined reference to `Sophus::SE3::operator*(Sophus::SE3 const&) const'
useSophus.cpp:(.text+0x67b): undefined reference to `Sophus::SE3::matrix() const'
CMakeFiles/useSophus.dir/useSophus.cpp.o: In function `Sophus::operator<<(std::ostream&, Sophus::SO3 const&)':
useSophus.cpp:(.text._ZN6SophuslsERSoRKNS_3SO3E[_ZN6SophuslsERSoRKNS_3SO3E]+0x2e): undefined reference to `Sophus::SO3::log() const'
collect2: error: ld returned 1 exit status
CMakeFiles/useSophus.dir/build.make:94: recipe for target 'useSophus' failed
make[2]: *** [useSophus] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/useSophus.dir/all' failed
make[1]: *** [CMakeFiles/useSophus.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

gaoxiang12, 2017-05-23:
请检查机器上是否有安装其他版本的sophus，包括可能位于其他项目中的。

robotdream, 2017-05-23:
谢谢，解决了。在CMakeLists.txt 里面加上set(Sophus_LIBRARIES libSophus.so)
如果找不到libSophus.so, 在make install sophus后用sudo ldconfig更新下

closed 2017-05-24 by gaoxiang12

LOGHORIZION, 2019-08-13:
CHANGE the 8th line of original code into #include <sophus/so3.cpp> will also work

tigerk0430, 2019-08-21:
> Thank you, solved. Add set(Sophus_LIBRARIES libSophus.so) to CMakeLists.txt.
> If libSophus.so is not found, update it with sudo ldconfig after make install sophus

It worked for me. Ch11 had a same isuse and solved it by following above instruction.