gaoxiang12/slambook #22: ch5/joinMap中VTK的问题

whyscience, 2017-06-05:
报错，但是依然可以编译通过
```
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
--   atomic
--   regex
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- looking for PCL_COMPONENT
-- Could NOT find PCL_COMPONENT (missing:  PCL_COMPONENT_LIBRARY PCL_COMPONENT_INCLUDE_DIR) 
-- looking for PCL_COMMON
-- looking for PCL_OCTREE
-- looking for PCL_IO
-- Configuring done
-- Generating done
-- Build files have been written to:
```

看  https://stackoverflow.com/questions/37369369/compiling-pcl-1-7-on-ubuntu-16-04-errors-in-cmake-generated-makefile
这里的提问好像是 vtk的问题，不知道有没有办法消除这个问题？

whyscience, 2017-06-05:
依然报错
环境是Ubuntu 16.04
```
~/D/s/c/joinMap> pcl_viewer map.pcd
The viewer window provides interactive commands; for help, press 'h' or 'H' from within the window.
> Loading map.pcd X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  154 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  43
  Current serial number in output stream:  44
```

whyscience, 2017-06-05:
上面的问题好像解决了 ： https://askubuntu.com/questions/893922/ubuntu-16-04-x-error-of-failed-request-badvalue-integer-parameter-out-of-range
NVIDIA驱动的问题

closed 2017-06-07 by gaoxiang12

jianrui1, 2019-01-11:
你好　-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist
这个问题是怎么解决的
为也遇到一样的问题，能否指教一下

zoumson, 2021-06-11:
change COMPONENT to COMPONENTS from CMake file. Add the `s`