gaoxiang12/slambook #27: 第九章Prject运行出现段错误，怎么解决？

simple1082, 2017-06-23:
Using host libthread_db library "/lib/i386-linux-gnu/libthread_db.so.1".
Core was generated by `bin/run_vo config/default.yaml'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0 0xb77a4501 in myslam::Frame::Frame(long, double, Sophus::SE3, std::shared_ptrmyslam::Camera, cv::Mat, cv::Mat) () from /home/dan/work/project/0.4/lib/libmyslam.so
(gdb) bt
#0 0xb77a4501 in myslam::Frame::Frame(long, double, Sophus::SE3, std::shared_ptrmyslam::Camera, cv::Mat, cv::Mat) () from /home/dan/work/project/0.4/lib/libmyslam.so
#1 0xb77a47ef in myslam::Frame::createFrame() () from /home/dan/work/project/0.4/lib/libmyslam.so
#2 0x0804b4d6 in main ()

lbwlele, 2017-07-05:
一模一样的问题，gdb定位一样也

lbwlele, 2017-08-19:
已解决，请检查自己的ubuntu版本，书中要求是ubuntu 14.04而不能是ubuntu kylin 14.04

zhanzhongwei, 2018-01-11:
修改函数Frame::Ptr Frame::createFrame()
将其中的
     “return Frame::Ptr( new Frame(factory_id++) );”
 修改为
“ Frame::Ptr framePtr = Frame::Ptr(new(std::nothrow) Frame());
if(framePtr != NULL){
framePtr->id_ = factory_id++;
return framePtr;
}“
修改后验证运行OK，其实就是new的时候传入初始化参数导致的问题，但是还没弄明白为什么这样会引起segmentation fault,如果有谁知道的话，可以分享下。

OUCwangxiang, 2018-06-11:
@zhanzhongwei is right!!!!!!

icylogic, 2018-10-11:
## Real Cause:

```
createFrame() =>
new Frame() =>
Frame.T_c_w_ (of type SE3d)
```

>If you define a structure having members of fixed-size vectorizable Eigen types, you must overload its "operator new" so that it generates 16-bytes-aligned pointers. Fortunately, Eigen provides you with a macro EIGEN_MAKE_ALIGNED_OPERATOR_NEW that does that for you.

more details: 
- http://eigen.tuxfamily.org/dox-devel/group__TopicStructHavingEigenMembers.html
- http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html

## Solution:
add `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` to `frame.h`:

```c++
class Frame
{
  ...
  SE3                            T_c_w_;      // transform from world to camera
  ...
public:
  ...
  EIGEN_MAKE_ALIGNED_OPERATOR_NEW
};
```
then clean and rebuild ( if you don't know how, just `rm -rf build/*` )

PS. How to locate this bug

1. change cmake configuration profile to Debug.
2. run the program. or use a debugger to get a more clear call stack

[image]

cc @zhanzhongwei