gaoxiang12/slambook #34: useSophus.cpp:8:24: fatal error: sophus/so3.h

zgq91, 2017-07-16:
高博你好：
   在实践Sophus中 报错fatal error: sophus/so3.h: No such file or directory #include "sophus/so3.h"

Sophus：
安装过程:
git checkout a621ff
mkdir build
cd build
cmake ..
make

useSophus.cpp的CMakeLists.txt   Sophus配置find_package( Sophus REQUIRED )
include_directories( ${Sophus_INCLUDE_DIRS} )

系统
ubuntu14.04
请问怎么解决这个错误。  电脑中配置过lsd_slam。

gaoxiang12, 2017-08-12:
检查下find_package为何没找到的原因。
或者手动指定sophus的头文件与库文件的目录。

ArmstrongWall, 2017-11-01:
在cmake配置中加上include_directories(  "/home/wzq/3rd_party/Sophus"  )
link_libraries("/home/wzq/3rd_party/Sophus/build/libSophus.so")
把目录换成你的

district10, 2018-03-18:
我也是不知道为啥今天 sophus 就挂了, 可以改成这样看看有没有配置对:

```cmake
find_package( Sophus REQUIRED )
include_directories( ${Sophus_INCLUDE_DIRS} )
IF( NOT Sophus_LIBRARIES )
    SET( Sophus_LIBRARIES "libSophus.so" )
ENDIF()
message( STATUS "Sophus library status:" )
message( STATUS "    includes: ${Sophus_INCLUDE_DIRS}" )
message( STATUS "    libraries: ${Sophus_LIBRARIES}" )
# -- Sophus library status:
# --     includes: /usr/local/lib/cmake/Sophus/../../../include
# --     libraries: libSophus.so
```

zhangxiaoya, 2018-03-21:
sophus是一个很小的库，不用安装，完全可以集成在工程中，find_package 是寻找SophusConfig.cmake文件，这里面说明了你的Sophus库安装在哪里，如果只是make了，没有make install，那把Sophus的目录就放在当前项目根目录中，这个工程中的find_package会在当前目录中找这个配置文件。

easonwang717, 2019-03-25:
> 高博你好：
> 在实践Sophus中 报错fatal error: sophus/so3.h: No such file or directory #include "sophus/so3.h"
>
> Sophus：
> 安装过程:
> git checkout a621ff
> mkdir build
> cd build
> cmake ..
> make
>
> useSophus.cpp的CMakeLists.txt [...]

同遇到这个问题，所以你最后是怎么解决的呢？

deeprun, 2021-03-13:
How did you solve this problem?