gaoxiang12/slambook #35: ch5 joinMap exit code 139

CNchence, 2017-08-01:
运行例程ch5 中的点云拼接程序，报错 
Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)

大概是溢出的问题，具体原因不得而知。
定位问题 到代码第36行：
poses.push_back( T );

修改意见：将poses定义为数列。
第15行： vector<Eigen::Isometry3d> poses;         // 相机位姿
修改为： Eigen::Isometry3d poses[5];

再运行 就成功了。
希望能帮助到和我遇到相同问题的人！

XZLeo, 2018-05-10:
我用之前下载的代码也遇到了同样的错误，刚入门c++，还不太熟悉语法，没有看您的修改意见，能把第15行完整代码给我看看吗？并且，我把code那里的代码沾过去，运行的结果变成了这样：

正在将图像转换为点云...
转换图像中: 1
转换图像中: 2
转换图像中: 3
转换图像中: 4
转换图像中: 5
点云共有0个点.
terminate called after throwing an instance of 'pcl::IOException'
  what():  : [pcl::PCDWriter::writeBinary] Input point cloud has no data!

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)

gs14iitbbs, 2018-07-02:
Its not a big issue; actually the path to images are not suited that is why it is not able to read the images and giving errors. So change the path accordingly.

I did this th following, but you can change accordingly.

boost::format fmt( "/home/gaurav/VO_and_SLAM_codes/slambook-master/ch5/joinMap/%s/%d.%s" ); //图像文件格式
        colorImgs.push_back( cv::imread( (fmt%"color"%(i+1)%"png").str() ));
        depthImgs.push_back( cv::imread( (fmt%"depth"%(i+1)%"pgm").str(), -1 )); // 使用-1读取原始图像

gaoxiang12, 2018-07-16:
the vector of Eigen objects should use the Eigen aligned allocators, see [https://eigen.tuxfamily.org/dox/group__TopicFixedSizeVectorizable.html](https://eigen.tuxfamily.org/dox/group__TopicFixedSizeVectorizable.html)

I'll fix it in the second edition.

voyage19941119, 2019-01-10:
> the vector of Eigen objects should use the Eigen aligned allocators, see https://eigen.tuxfamily.org/dox/group__TopicFixedSizeVectorizable.html
> 
> I'll fix it in the second edition.

高博，这种问题我在其他地方也多次遇到过，看起来是把Eigen类型作为vector的元素导致的内存错误。而且把Eigen类型作为函数的传入传出参数也容易导致内存问题。