gaoxiang12/slambook #38: ch4/useSophus Error

teduav, 2017-08-23:
~/slambook/ch4/useSophus$ make
Scanning dependencies of target useSophus
[ 50%] Building CXX object CMakeFiles/useSophus.dir/useSophus.cpp.o
[100%] Linking CXX executable useSophus
CMakeFiles/useSophus.dir/useSophus.cpp.o：在函数‘main’中：
useSophus.cpp:(.text+0xd1)：对‘Sophus::SO3::SO3(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&)’未定义的引用
useSophus.cpp:(.text+0xf4)：对‘Sophus::SO3::SO3(double, double, double)’未定义的引用
useSophus.cpp:(.text+0x126)：对‘Sophus::SO3::SO3(Eigen::Quaternion<double, 0> const&)’未定义的引用
useSophus.cpp:(.text+0x1d2)：对‘Sophus::SO3::log() const’未定义的引用
useSophus.cpp:(.text+0x232)：对‘Sophus::SO3::hat(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x27c)：对‘Sophus::SO3::hat(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x295)：对‘Sophus::SO3::vee(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&)’未定义的引用
useSophus.cpp:(.text+0x341)：对‘Sophus::SO3::exp(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x361)：对‘Sophus::SO3::operator*(Sophus::SO3 const&) const’未定义的引用
useSophus.cpp:(.text+0x41a)：对‘Sophus::SE3::SE3(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x43a)：对‘Sophus::SE3::SE3(Eigen::Quaternion<double, 0> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x4cf)：对‘Sophus::SE3::log() const’未定义的引用
useSophus.cpp:(.text+0x52f)：对‘Sophus::SE3::hat(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x586)：对‘Sophus::SE3::hat(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x59f)：对‘Sophus::SE3::vee(Eigen::Matrix<double, 4, 4, 0, 4, 4> const&)’未定义的引用
useSophus.cpp:(.text+0x642)：对‘Sophus::SE3::exp(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x662)：对‘Sophus::SE3::operator*(Sophus::SE3 const&) const’未定义的引用
useSophus.cpp:(.text+0x67b)：对‘Sophus::SE3::matrix() const’未定义的引用
CMakeFiles/useSophus.dir/useSophus.cpp.o：在函数‘Sophus::operator<<(std::ostream&, Sophus::SO3 const&)’中：
useSophus.cpp:(.text._ZN6SophuslsERSoRKNS_3SO3E[_ZN6SophuslsERSoRKNS_3SO3E]+0x2e)：对‘Sophus::SO3::log() const’未定义的引用
collect2: error: ld returned 1 exit status
CMakeFiles/useSophus.dir/build.make:94: recipe for target 'useSophus' failed
make[2]: *** [useSophus] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/useSophus.dir/all' failed
make[1]: *** [CMakeFiles/useSophus.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

teduav, 2017-08-23:
编译UseSophus.cpp时出错，希望大家帮忙解答下，谢谢。

gaoxiang12, 2017-09-01:
没有链接到sophus库。

ArmstrongWall, 2017-11-01:
手动指定头文件目录，手动添加动态链接库
include_directories(  /home/wzq/3rd_party/Sophus  )
target_link_libraries(project libSophus.so)

ArmstrongWall, 2017-11-01:
还有link_libraries("/home/wzq/3rd_party/Sophus/build/libSophus.so")

wwpokenet, 2017-12-27:
当是target_link_libraries( useSophus ${Sophus_LIBRARIES})时
我的${Sophus_LIBRARIES}下的确有libSophus.dylib这个库，为什么编译仍然不能通过
只有改为target_link_libraries( useSophus libSophus.dylib)才能通过。
求教🤔🤔🤔

electech6, 2018-03-14:
我的也是这样：将
target_link_libraries( useSophus ${Sophus_LIBRARIES} )
改成这样才OK
target_link_libraries(useSophus /usr/local/lib/libSophus.so)
不太明白为啥find_package 找到的Sophus_LIBRARIES不行

Amazingren, 2018-08-10:
解决了吗？

roachsinai, 2018-11-29:
这个Sophus能让cmake一些配置命令无效也是神奇，不知道那里的原因。