gaoxiang12/slambook #41: ch3.visualizeGeometry

ZzJiachenchen, 2017-09-02:
可爱的高博士您好，我在VMstation上运行visualizeGeometry程序的时候，
也遇到了报错“what(): Pangolin X11: Unable to retrieve framebuffer options”，
但没看懂...paulinus 提出的解决办法...
是要修改Pangolin/src/display/device/display_x11.cpp文件中，paulinus标记的123行处的注释内容么？
期待着您的回复，十万分谢谢您！

SiuKeungm, 2017-09-03:
[image]

SiuKeungm, 2017-09-03:
构建时候出现了下面的error。不知道如何解决
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:67: undefined reference to `pangolin::CreateWindowAndBind(std::string, int, int, pangolin::Params const&)'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:68: undefined reference to `glEnable'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:72: undefined reference to `pangolin::ModelViewLookAt(double, double, double, double, double, double, pangolin::AxisDirection)'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:72: undefined reference to `pangolin::ProjectionMatrix(int, int, double, double, double, double, double, double)'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:72: undefined reference to `pangolin::OpenGlRenderState::OpenGlRenderState(pangolin::OpenGlMatrix const&, pangolin::OpenGlMatrix const&)'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:76: undefined reference to `pangolin::Handler3D::Handler3D(pangolin::OpenGlRenderState&, pangolin::AxisDirection, float, float)'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:76: undefined reference to `pangolin::CreateDisplay()'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:76: undefined reference to `pangolin::View::SetBounds(pangolin::Attach, pangolin::Attach, pangolin::Attach, pangolin::Attach, double)'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:76: undefined reference to `pangolin::View::SetHandler(pangolin::Handler*)'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:83: undefined reference to `pangolin::CreatePanel(std::string const&)'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:84: undefined reference to `pangolin::View::SetBounds(pangolin::Attach, pangolin::Attach, pangolin::Attach, pangolin::Attach)'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:88: undefined reference to `glClear'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:90: undefined reference to `pangolin::View::Activate(pangolin::OpenGlRenderState const&) const'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:92: undefined reference to `pangolin::OpenGlRenderState::GetModelViewMatrix()'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:114: undefined reference to `glColor3f'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:118: undefined reference to `glLineWidth'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:119: undefined reference to `glColor3f'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:120: undefined reference to `glBegin'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:121: undefined reference to `glVertex3f'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:122: undefined reference to `glVertex3f'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:123: undefined reference to `glColor3f'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:124: undefined reference to `glVertex3f'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:125: undefined reference to `glVertex3f'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:126: undefined reference to `glColor3f'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:127: undefined reference to `glVertex3f'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:128: undefined reference to `glVertex3f'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:129: undefined reference to `glEnd'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:131: undefined reference to `pangolin::FinishFrame()'
/home/siukeung/slambook1/ch3/visualizeGeometry/visualizeGeometry.cpp:86: undefined reference to `pangolin::ShouldQuit()'
CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o:(.data+0x0): undefined reference to `vtable for pangolin::Handler'
CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o:(.data+0x8): undefined reference to `vtable for pangolin::HandlerScroll'
CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o: In function `pangolin::_CheckGlDieOnError(char const*, int)':
/usr/local/include/pangolin/gl/glinclude.h:40: undefined reference to `glGetError'
/usr/local/include/pangolin/gl/glinclude.h:42: undefined reference to `pangolin::glErrorString(unsigned int)'
CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o: In function `pangolin::GlTexture::~GlTexture()':
/usr/local/include/pangolin/gl/gl.hpp:149: undefined reference to `pangolin::ShouldQuit()'
/usr/local/include/pangolin/gl/gl.hpp:150: undefined reference to `glDeleteTextures'
CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o: In function `pangolin::GlTexture::Bind() const':
/usr/local/include/pangolin/gl/gl.hpp:156: undefined reference to `glBindTexture'
CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o: In function `pangolin::GlTexture::Reinitialise(int, int, int, bool, int, unsigned int, unsigned int, void*)':
/usr/local/include/pangolin/gl/gl.hpp:167: undefined reference to `glDeleteTextures'
/usr/local/include/pangolin/gl/gl.hpp:174: undefined reference to `glGenTextures'
/usr/local/include/pangolin/gl/gl.hpp:179: undefined reference to `glTexImage2D'
[... cut ...]
/usr/local/include/pangolin/var/var.h:79: undefined reference to `pangolin::VarState::I()'
CMakeFiles/visualizeGeometry.dir/visualizeGeometry.cpp.o: In function `pangolin::HandlerScroll::~HandlerScroll()':
/usr/local/include/pangolin/handler/handler.h:65: undefined reference to `vtable for pangolin::HandlerScroll'
collect2: error: ld returned 1 exit status
make[2]: *** [visualizeGeometry] Error 1
make[1]: *** [CMakeFiles/visualizeGeometry.dir/all] Error 2
make: *** [all] Error 2
*** 失败：退出代码 2 ***

SiuKeungm, 2017-09-03:
在KDEVELOPE里面编译会出错，但放到终端里面cmake又正常了，不知道为啥

amlearner, 2018-07-31:
注释掉123和124两行。[https://github.com/stevenlovegrove/Pangolin/commit/45b38c15fae075aec0e3e865780b5235b4055848](url)

gaoxiang12, 2018-08-01:
是的，地址见楼上，在虚拟机内可能出现这种情况。
问题解决的话请关闭这个issue，祝好。

closed 2019-06-04 by ZzJiachenchen