gaoxiang12/slambook #46: 运行第七章pose_estimation_3d2d程序时终端提示double free or corruption(out)

FinleyPan, 2017-09-11:
我的系统是ubuntu16.04 ,初步判定问题出在bundle adjustment那个函数里面产生了内存泄露，但当我把相机参数对象的初始化改为默认构造函数后：
`g2o::CameraParameters* camera = new g2o::CameraParameters()`
  程序就可以运行了，但是输出最终的位姿T后终端会提示段错误。不知道有人遇到跟我一样的问题吗？

FinleyPan, 2017-09-11:
g2o更换为高博书中提供的g2o，重新编译安装后就跑通了。是g2o版本问题？

Jaypancool, 2017-12-28:
应该是g2o库的版本问题，你的电脑里面是不是安装了很多g2o的库？ 在CMakeLists.txt文件里面使用：
include_directories(
    "/usr/local/include/g2o"
    "/usr/include/eigen3"
    )
写死调用g2o库的位置，可以解决这个问题。
具体原因，应该和系统调用库的默认路径有关。

gcjGithub, 2018-03-11:
@Jaypancool [请问我在这下载的源码直接安装g2o。](https://github.com/RainerKuemmerle/g2o) 我没有安装高博书上的g2o，我在3d-2d除了修改了优化器那部分外 其他地方都没有修改 运行时出现double free错误，当我定位错误的时候 ，出现在g2o的内部库函数里面。 @FinleyPan 你的那种错误我的也是出现了，到现在还没有解决。我按照jaypancool提供的方法 包含头文件的时候直接写死调用g2o库位置，但是仍然没有解决。请问有没有更好的解决方案 谢谢