gaoxiang12/slambook #47: 关于第5讲操作Opencv图像中头文件引入的两个小疑问

Harry-Zhi, 2017-09-12:
我们在imageBasics.cpp中包含了opencv2的两个头文件：
```
#include <opencv2/core/core.hpp>
#include<opencv2/highgui/highgui.hpp>
```
好奇的打开了这两个文件，发现他们分别引入的是上一级目录opencv2根目录下的core.hpp和highgui.hpp,
例如opencv2/core/core.hpp文件中内容只有：
```

#ifdef __OPENCV_BUILD
#error this is a compatibility header which should not be used inside the OpenCV library
#endif

#include "opencv2/highgui.hpp"
```
想请问高博，我们实际中，以及本例中采用这么一种调用方式而不直接#include <opencv2/core.hpp>的原因是什么呢？
另外，代码中说this is a compatibility header which should not be used inside the OpenCV library，是指我们不该用这个程序么？

麻烦您解答一下我的小疑惑，谢谢您了。

tyuownu, 2018-03-17:
我的不是啊。我的里面内容很多的，一定是为了兼容，不要纠结太多。

zhangxiaoya, 2018-03-21:
@Harry-Zhi core头文件是包含opencv库的基本数据结构的定义，以及其他的一些基本的算法，higui头文件是为了引入higui模块，是high level GUI的简写，主要是与UI显示相关的东西，imshow函数就是定义在这个模块，ImageBasic文件中对Mat的基本操作core模块就足够了，显示结果会用来higui。[OpenCV各个模块的定义](https://docs.opencv.org/3.1.0/)