gaoxiang12/slambook #48: 在使用Sophus的程序中，为何CmakeLists没有显示引用Eigen的路径就能顺利编译呢？

Harry-Zhi, 2017-09-12:
高博您好，想请问下Sophus的操作中，源代码中有#include<Eigen/Core>。

但为何CmakeLists中未引入Eigen的路径就能顺利编译成功了呢？

从原理上我觉得是否要显示的引入所有第三库的路径？（除非库的头文件在系统PATH里）

麻烦您抽空解答一下。

谢谢您了。

小智

CmakeLists：
```
cmake_minimum_required( VERSION 2.8 )
project( useSophus )

# 为使用 sophus，您需要使用find_package命令找到它
find_package( Sophus REQUIRED )
include_directories( ${Sophus_INCLUDE_DIRS} )

add_executable( useSophus useSophus.cpp )
target_link_libraries( useSophus ${Sophus_LIBRARIES} )
```

YuehChuan, 2017-09-13:
@Harry-Zhi Eigen標頭檔內建在/usr/local/include , 假使您之前裝過eigen

Harry-Zhi, 2017-09-13:
@YuehChuan 明白了，谢谢你的帮助。

此外，想问下那这是否代表，在ch3练习中，我们在cmakelists显示加入的：
`include_directories( "/usr/include/eigen3" )`，
这句话其实是可以忽略的。

也许当时只是为了强调路径的重要性加入了这句话？

YuehChuan, 2017-09-13:
哦 抱歉 沒看清楚 eigen 安裝位置是  /usr/include/eigen3
所以在編譯的時候要加  標頭檔位置 
sophus 不用加因為sophus函式庫編譯時候include eigen3
[image]

不過這下換我困惑的是sophus為什麼不用make install安裝   find package就能找的到, 另外#include<>
尖括弧 不是會去自動去找 /usr/local/include  /usr/include   Mmmmmm

YuehChuan, 2017-09-13:
[image]
是sophus CMakeLists.txt這裡呢  真神奇

YuehChuan, 2017-09-13:
[image]

YuehChuan, 2017-09-13:
https://cmake.org/cmake/help/v3.0/command/export.html

likehuaer, 2019-04-21:
我这边的情况是：
1. 编译sophus之前已经安装了eigen3，没有make install sophus，sophus的头文件是找不到的。
2. 让我疑惑的sophus的头文件找到了以后，eigen3的头文件为什么还是链接不到呢？非得再CMakelists.txt加上eigen3的头文件。
[image]