gaoxiang12/slambook #49: ch5点云拼接auto& d:data编译出错

ZzJiachenchen, 2017-09-16:
有遇到如下编译报错的小伙伴么，还在学C++，学的不够多，不知如何改正，求指点，谢谢你们！
/home/zjc19941015/slambook/ch55/joinMap.cpp:31:12: error: ISO C++ forbids declaration of ‘d’ with no type [-fpermissive]
 for (auto& d:data)
            ^
/home/zjc19941015/slambook/ch55/joinMap.cpp:31:14: error: range-based ‘for’ loops are not allowed in C++98 mode
 for (auto& d:data)
              ^

shaozu, 2017-09-17:
是不是编译的时候没有加支持c++11的选项。
试着在CMakeLists.txt中加上`set( CMAKE_CXX_FLAGS "-std=c++11 -O3" )`。

ZzJiachenchen, 2017-09-20:
嗯嗯是的，第一次得到网友解答问题，哈哈哈谢谢小天使！
不过在CMake中加入您写出的一行代码报错，加的是同样功能的一句代码：
`add_compile_options(-std=c++11)`

happyrookie, 2018-11-09:
for (auto& d:data)，楼主这个语句是什么意思呢？

shaozu, 2018-11-09:
@happyrookie 参考 https://zh.cppreference.com/w/cpp/language/range-for

closed 2019-06-04 by ZzJiachenchen