gaoxiang12/slambook #52: ch6/g2o_curve_fitting/CMakeLists.txt issue

game-of-drones, 2017-09-20:
After checking cmake_modules/FindG2O.cmake, I think the following part of the CMakeLists.txt of g2o_curve_fitting should be adjusted:
1. `${G2O_INCLUDE_DIRS}` should be `${G2O_INCLUDE_DIR}`. 
2.
```
target_link_libraries( curve_fitting 
    ${OpenCV_LIBS}
    g2o_core g2o_stuff
)
```
should be
```
target_link_libraries( curve_fitting 
    ${OpenCV_LIBS}
    ${G2O_CORE_LIBRARY} ${G2O_STUFF_LIBRARY}
)
```
Without 1, I can still build the code without problem, which is surprising.
Wihtout 2, I got the following error when run `curve_fitting`
> error while loading shared libraries: libg2o_core.so: cannot open shared object file: No such file or directory

hgkdzbf6, 2017-11-20:
thank you! i solved my errors!

wodemaa5, 2018-01-23:
Yes,u are right . It's helping .thank u !

hanzheteng, 2020-03-25:
From https://github.com/gaoxiang12/slambook/issues/125#issuecomment-603575294, to temporarily add the search path, the following way also works:
```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib
```