gaoxiang12/slambook #61: 第七章pose_estimation_3d3d

Harry-Zhi, 2017-11-13:
高博您好，
在您第七章pose_estimation_3d3d 代码的295行，
edge->setInformation( Eigen::Matrix3d::Identity()*1e4 );
为何这里要给信息矩阵乘以1e4;
而在3d2d（也即PNP）的代码中，信息矩阵没有乘以这个系数。

不过就算去掉这里的1e4实验结果结果也不变，不晓得这里加这个系数有什么insights?

mengkaitao, 2018-02-05:
出现错误，ch7的pose_estimation_3d3d编译出错，跟据网上各种方法没有解决，十分苦恼，还请大神解惑

[ 10%] Building CXX object CMakeFiles/pose_estimation_3d3d.dir/pose_estimation_3d3d.cpp.o
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp: In function ‘void bundleAdjustment(const std::vector<cv::Point3_<float> >&, const std::vector<cv::Point3_<float> >&, cv::Mat&, cv::Mat&)’:
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:269:54: error: ‘LinearSolverCSparse’ in namespace ‘g2o’ does not name a template type
     Block::LinearSolverType* linearSolver = new g2o::LinearSolverCSpa
                                                      ^
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:269:95: error: expected primary-expression before ‘>’ token
 olver = new g2o::LinearSolverCSparse<Block::PoseMatrixType>(); // 线
                                                           ^
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:269:97: error: expected primary-expression before ‘)’ token
 ver = new g2o::LinearSolverCSparse<Block::PoseMatrixType>(); // 线��
                                                           ^
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:270:50: error: no matching function for call to ‘g2o::BlockSolver<g2o::BlockSolverTraits<6, 3> >::BlockSolver(g2o::BlockSolver<g2o::BlockSolverTraits<6, 3> >::LinearSolverType*&)’
     Block* solver_ptr = new Block ( linearSolver );     // 矩阵块�
                                                  ^
In file included from /usr/local/include/g2o/core/block_solver.h:199:0,
                 from /home/meng/slambook/ch7/pose_estimation_3d3d.cpp:11:
/usr/local/include/g2o/core/block_solver.hpp:40:1: note: candidate: g2o::BlockSolver<Traits>::BlockSolver(std::unique_ptr<typename Traits::LinearSolverType>) [with Traits = g2o::BlockSolverTraits<6, 3>; typename Traits::LinearSolverType = g2o::LinearSolver<Eigen::Matrix<double, 6, 6, 0> >]
 BlockSolver<Traits>::BlockSolver(std::unique_ptr<LinearSolverType> li
 ^
/usr/local/include/g2o/core/block_solver.hpp:40:1: note:   no known conversion for argument 1 from ‘g2o::BlockSolver<g2o::BlockSolverTraits<6, 3> >::LinearSolverType* {aka g2o::LinearSolver<Eigen::Matrix<double, 6, 6, 0> >*}’ to ‘std::unique_ptr<g2o::LinearSolver<Eigen::Matrix<double, 6, 6, 0> >, std::default_delete<g2o::LinearSolver<Eigen::Matrix<double, 6, 6, 0> > > >’
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:271:5: error: ‘OptimizationAlgorithmLevenberg’ is not a member of ‘g2o’
     g2o::OptimizationAlgorithmLevenberg* solver = new g2o::Optimizati
     ^
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:271:42: error: ‘solver’ was not declared in this scope
     g2o::OptimizationAlgorithmLevenberg* solver = new g2o::Optimizati
                                          ^
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:271:55: error: expected type-specifier
     g2o::OptimizationAlgorithmLevenberg* solver = new g2o::Optimizati
                                                       ^
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:315:66: error: ‘pose’ was not declared in this scope
     edge->setVertex( 0, dynamic_cast<g2o::VertexSE3Expmap*> (pose) );
                                                              ^
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:333:42: error: ‘pose’ was not declared in this scope
     cout<<"T="<<endl<<Eigen::Isometry3d( pose->estimate() ).matrix()<

zhangxiaoya, 2018-03-21:
@mengkaitao 你这是g2o版本有问题吧，你看一下g2o的这个[Issue206](https://github.com/RainerKuemmerle/g2o/issues/206)，作者给出可解决方法，切换老的版本就行了