gaoxiang12/slambook #65: ch5编译问题

zhouweijlu, 2017-11-27:
这是cmake之后的
Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
--   atomic
--   regex
-- Checking for module 'libopenni2'
--   No package 'libopenni2' found
-- Could NOT find OpenNI2 (missing:  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- looking for PCL_COMPONENT
-- Could NOT find PCL_COMPONENT (missing:  PCL_COMPONENT_LIBRARY PCL_COMPONENT_INCLUDE_DIR) 
-- looking for PCL_COMMON
-- looking for PCL_OCTREE
-- looking for PCL_IO
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wolverine/exercise/ch5/joinMap
编译是可以通过的，可执行文件也可以运行，但是会报错，
Loading map.pcd ERROR: In /build/vtk6-dmAaMa/vtk6-6.2.0+dfsg1/Rendering/OpenGL/vtkOpenGLTexture.cxx, line 200
vtkOpenGLTexture (0x1e644e0): No scalar values found for texture input!
有遇到类似问题的吗？

zwlovefish, 2018-02-28:
同求，ERROR: In /build/vtk6-YpT4yb/vtk6-6.2.0+dfsg1/Rendering/OpenGL/vtkOpenGLTexture.cxx, line 200

dongmingsun, 2018-04-12:
更新下`VTK`版本呢？

szuer2017, 2018-04-23:
Have you addressed this problem ,xiongdei?

ChurchKing, 2018-09-08:
和你的一模一样,解决没?

szuer2017, 2018-09-15:
> 和你的一模一样,解决没?

我已经忘了问题，但是愿意跟你分享一下思路；
    原博客里面的问题是：
            Could NOT find PCL_COMPONENT (missing: PCL_COMPONENT_LIBRARY PCL_COMPONENT_INCLUDE_DIR)
 建议重新安装一下PCL_COMPONENT ,再试一下编译。

calmelo, 2020-07-14:
所以最后解决办法是什么？

zoumson, 2021-06-11:
change COMPONENT to COMPONENTS from cmake file

damagebro, 2022-10-10:
我使用ubuntu18.04编译pcl点云库，也遇到这个问题，可以尝试这样解决：

*书本中原始安装pcl库的命令:
sudo apt-get install libpcl-all
*替换为：
sudo apt-get install libpcl-dev pcl-tools