gaoxiang12/slambook #75: 第六章编译提示错误

xiaoyaoyouhuanle, 2018-01-17:
By not providing "FindCeres.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Ceres", but
  CMake did not find one.

  Could not find a package configuration file provided by "Ceres" with any of
  the following names:

    CeresConfig.cmake
    ceres-config.cmake
请问如何解决？

gaoxiang12, 2018-01-17:
找一个FindCeres.cmake的文件，加到cmake的搜索列表里。
命令可以自己查。

huntsmanydw, 2020-10-20:
I also fund this error, in windows platform.
But I can not run sudo make install,
how to solve this error?

sirrusyqh, 2020-12-08:
my solution:
1、find the ceres directory. such as:  /home/ceres-solver
2、enter the "build" direcotory and execute cmake  .. and make and sudo make install
3、if you meet No opencv directory , find the according directory and append it in your CMakeLists.txt. such as
set(OpenCV_DIR /home/johnny/opencv-3.4.3/build)