gaoxiang12/slambook #76: 第九章project/0.3 optimizer.optimize ( 10 )报错

wwpokenet, 2018-01-19:
在运行第九章project/0.3/bin/run_vo以及/project/0.4/bin/run_vo时，一旦程序调用optimizer.optimize ( 10 )就会报错，0.3版本会分提示：
Assertion failed: (( ((internal::UIntPtr(m_data) % internal::traits<Derived>::Alignment) == 0) || (cols() * rows() * innerStride() * sizeof(Scalar)) < internal::traits<Derived>::Alignment ) && "data is not aligned"), function checkSanity, file /usr/local/lib/cmake/Sophus/../../../include/Eigen/src/Core/MapBase.h, line 191.
Abort trap: 6
或者：
run_vo(39088,0x7fff9f0a7340) malloc: *** mach_vm_map(size=9289677391957696512) failed (error code=3)
*** error: can't allocate region
*** set a breakpoint in malloc_error_break to debug
libc++abi.dylib: terminating with uncaught exception of type std::bad_alloc: std::bad_alloc
Abort trap: 6
0.4版本提示为：
run_vo(39086,0x7fff9f0a7340) malloc: *** mach_vm_map(size=9289450251538436096) failed (error code=3)
*** error: can't allocate region
*** set a breakpoint in malloc_error_break to debug
libc++abi.dylib: terminating with uncaught exception of type std::bad_alloc: std::bad_alloc
Abort trap: 6

matheecs, 2018-05-01:
内存对齐的问题，建议修改CMakeLists 为
` set( CMAKE_CXX_FLAGS "-std=c++11" )`

roachsinai, 2018-12-05:
问一下，第8章有遇到`std::bad_alloc`问题么？

IvyGongoogle, 2019-08-09:
modify `set( CMAKE_CXX_FLAGS "-std=c++11 -march=native -O3" )` to `set( CMAKE_CXX_FLAGS "-std=c++11" )` for macos in CMakeLists.txt