gaoxiang12/slambook #86: 第七章：pose_estimation 编译报错

mengkaitao, 2018-02-07:
ch7的pose_estimation_3d3d编译出错，跟据网上各种方法没有解决，十分苦恼，还请大神解惑
具体错误代码部分如下：
/*    typedef g2o::BlockSolver< g2o::BlockSolverTraits<6, 3> > Block; // pose 维度为 6, landmark 维度为 3
    Block::LinearSolverType* linearSolver = new g2o::LinearSolverCSparse<Block::PoseMatrixType>(); // 线性方程求解器
    Block* solver_ptr = new Block ( linearSolver );     // 矩阵块求解器
    g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg ( solver_ptr );
    g2o::SparseOptimizer optimizer;
    optimizer.setAlgorithm ( solver );
*/
    typedef g2o::BlockSolver< g2o::BlockSolverTraits<6,3> > Block;
    std::unique_ptr<Block::LinearSolverType> linearSolver ( new g2o::LinearSolverDense<Block::PoseMatrixType>() );
    std::unique_ptr<Block> solver_ptr (new Block( std::move(linearSolver) ));
    g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg( std::move(solver_ptr) );
    g2o::SparseOptimizer optimizer;
    optimizer.setAlgorithm( solver );

编译错误显示：

[ 10%] Building CXX object CMakeFiles/pose_estimation_3d3d.dir/pose_estimation_3d3d.cpp.o
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp: In function ‘void bundleAdjustment(const std::vector<cv::Point3_ >&, const std::vector<cv::Point3_ >&, cv::Mat&, cv::Mat&)’:
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:269:54: error: ‘LinearSolverCSparse’ in namespace ‘g2o’ does not name a template type
Block::LinearSolverType* linearSolver = new g2o::LinearSolverCSpa
^
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:269:95: error: expected primary-expression before ‘>’ token
olver = new g2o::LinearSolverCSparseBlock::PoseMatrixType(); // 线
^
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:269:97: error: expected primary-expression before ‘)’ token
ver = new g2o::LinearSolverCSparseBlock::PoseMatrixType(); // 线��
^
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:270:50: error: no matching function for call to ‘g2o::BlockSolver<g2o::BlockSolverTraits<6, 3> >::BlockSolver(g2o::BlockSolver<g2o::BlockSolverTraits<6, 3> >::LinearSolverType*&)’
Block* solver_ptr = new Block ( linearSolver ); // 矩阵块�
^
In file included from /usr/local/include/g2o/core/block_solver.h:199:0,
from /home/meng/slambook/ch7/pose_estimation_3d3d.cpp:11:
/usr/local/include/g2o/core/block_solver.hpp:40:1: note: candidate: g2o::BlockSolver::BlockSolver(std::unique_ptr) [with Traits = g2o::BlockSolverTraits<6, 3>; typename Traits::LinearSolverType = g2o::LinearSolver<Eigen::Matrix<double, 6, 6, 0> >]
BlockSolver::BlockSolver(std::unique_ptr li
^
/usr/local/include/g2o/core/block_solver.hpp:40:1: note: no known conversion for argument 1 from ‘g2o::BlockSolver<g2o::BlockSolverTraits<6, 3> >::LinearSolverType* {aka g2o::LinearSolver<Eigen::Matrix<double, 6, 6, 0> >}’ to ‘std::unique_ptr<g2o::LinearSolver<Eigen::Matrix<double, 6, 6, 0> >, std::default_delete<g2o::LinearSolver<Eigen::Matrix<double, 6, 6, 0> > > >’
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:271:5: error: ‘OptimizationAlgorithmLevenberg’ is not a member of ‘g2o’
g2o::OptimizationAlgorithmLevenberg solver = new g2o::Optimizati
^
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:271:42: error: ‘solver’ was not declared in this scope
g2o::OptimizationAlgorithmLevenberg* solver = new g2o::Optimizati
^
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:271:55: error: expected type-specifier
g2o::OptimizationAlgorithmLevenberg* solver = new g2o::Optimizati
^
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:315:66: error: ‘pose’ was not declared in this scope
edge->setVertex( 0, dynamic_castg2o::VertexSE3Expmap* (pose) );
^
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:333:42: error: ‘pose’ was not declared in this scope
cout<<"T="<<endl<<Eigen::Isometry3d( pose->estimate() ).matrix()<

gaoxiang12, 2018-02-07:
请使用本仓库提供的g2o版本。g2o最近更新过，语法转到了c++11，与旧版不兼容。

mengkaitao, 2018-02-07:
我按照您的提示安装了仓库提供的g2o，可是第七章的还是不行，第六章的反而原本书上的可以了，还是报错：
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:269:54: error: ‘LinearSolverCSparse’ in namespace ‘g2o’ does not name a template type
Block::LinearSolverType* linearSolver = new g2o::LinearSolverCSpa
^
/home/meng/slambook/ch7/pose_estimation_3d3d.cpp:269:95: error: expected primary-expression before ‘>’ token
olver = new g2o::LinearSolverCSparseBlock::PoseMatrixType(); // 线

十分苦恼，我的系统是ubuntu16.04，g2o的依赖库也都安装了，g2o的工程编译安装成功了，第六章和第七章对于g20的应用比较类似，但是为什么就是编译老师出错呀，被卡在这里好难受，还望指教

xr0604, 2020-08-14:
我也遇到类似的问题，请问楼主解决了吗