gaoxiang12/slambook #87: Ch7: feature_extraction drawMatches and imshow issues

gh279760559, 2018-03-01:
你好,
在尝试第七章feature 代码的时候发生如下问题:
[image]
无法显示matching points, 图片显示全黑.
已安装的opencv version = 3.3.1
following the link to install: https://www.learnopencv.com/install-opencv3-on-ubuntu/
谢谢

gh279760559, 2018-03-01:
update: 尝试显示一张图没有任何问题
[image]

district10, 2018-03-17:
@gh279760559 同样问题, 把 imshow 窗口的 title 换成英文就好了...

gh279760559, 2018-03-19:
@district10 多谢!

cdlwhm1217096231, 2018-03-20:
@gh279760559 请问怎么运行这个程序？我的编译通过了，但是不知道怎么运行，谢谢！

cdlwhm1217096231, 2018-03-20:
@district10 请问具体改哪里？谢谢！

district10, 2018-03-20:
@cdlwhm1217096231 把中文字符串换成英文的, 这两行: <https://github.com/gaoxiang12/slambook/blob/master/ch7/feature_extraction.cpp#L79-L80>

zhangxiaoya, 2018-03-21:
@gh279760559 你的系统是不是缺少中文语言包

gh279760559, 2018-03-21:
@zhangxiaoya 应该是,我在国外,并且使用的ubuntu 16.04lts系统

Krasaa, 2018-07-11:
@district10 多谢！

closed 2018-07-14 by gaoxiang12

shannon112, 2019-06-02:
感谢！ @district10