gaoxiang12/slambook #88: ch7pose3d2d编译报错

zwlovefish, 2018-03-04:
Scanning dependencies of target pose3d2d
[ 50%] Building CXX object CMakeFiles/pose3d2d.dir/main.cpp.o
[100%] Linking CXX executable pose3d2d
CMakeFiles/pose3d2d.dir/main.cpp.o: In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solvePattern(g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> >&, std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > > const&, g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&)':
main.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE12solvePatternERNS_17SparseBlockMatrixINS2_IdLin1ELin1ELi0ELin1ELin1EEEEERKSt6vectorISt4pairIiiESaISB_EERKNS5_IS3_EE[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE12solvePatternERNS_17SparseBlockMatrixINS2_IdLin1ELin1ELi0ELin1ELin1EEEEERKSt6vectorISt4pairIiiESaISB_EERKNS5_IS3_EE]+0xd8): undefined reference to `g2o::csparse_extension::cs_chol_workspace(cs_di_sparse const*, cs_di_symbolic const*, int*, double*)'
CMakeFiles/pose3d2d.dir/main.cpp.o: In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solveBlocks(double**&, g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&)':
main.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE11solveBlocksERPPdRKNS_17SparseBlockMatrixIS3_EE[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE11solveBlocksERPPdRKNS_17SparseBlockMatrixIS3_EE]+0xfd): undefined reference to `g2o::csparse_extension::cs_chol_workspace(cs_di_sparse const*, cs_di_symbolic const*, int*, double*)'
CMakeFiles/pose3d2d.dir/main.cpp.o: In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solve(g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&, double*, double*)':
main.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0xe8): undefined reference to `g2o::csparse_extension::cs_cholsolsymb(cs_di_sparse const*, double*, cs_di_symbolic const*, double*, int*)'
main.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0x1ea): undefined reference to `g2o::csparse_extension::writeCs2Octave(char const*, cs_di_sparse const*, bool)'
collect2: error: ld returned 1 exit status
CMakeFiles/pose3d2d.dir/build.make:112: recipe for target 'pose3d2d' failed
make[2]: *** [pose3d2d] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/pose3d2d.dir/all' failed
make[1]: *** [CMakeFiles/pose3d2d.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
*** Failure: Exit code 2 ***

sunjiayu12, 2018-03-05:
使用3rdparty中提供的g2o安装包，或者参照[g2o interface change ,old one not use](https://github.com/gaoxiang12/slambook/pull/85/files)修改代码。

wangtuo0820, 2018-03-09:
我也遇到了同样的问题
qmake -o Makefile ../pose_estimation_3d2d/pose_estimation_3d2d.pro -spec linux-g++
g++ -Wl,-O1 -o pose_estimation_3d2d main.o   /usr/local/lib/libopencv* /usr/lib/libceres.so /usr/local/lib/libg2o* /usr/lib/x86_64-linux-gnu/libcsparse* /usr/lib/x86_64-linux-gnu/libcxsparse* /usr/lib/x86_64-linux-gnu/libcholmod* /usr/lib/x86_64-linux-gnu/libglog.so /usr/lib/x86_64-linux-gnu/libwinpr-dsparse* /usr/lib/x86_64-linux-gnu/libsuitesparseconfig* 
我把和sparse有关的动态库全都链接上了，求大神帮助
`main.o:` In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solvePattern(g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> >&, std::vector<std::pair<int, int>, std::allocator<std::pair<int, int> > > const&, g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&)':
main.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE12solvePatternERNS_17SparseBlockMatrixINS2_IdLin1ELin1ELi0ELin1ELin1EEEEERKSt6vectorISt4pairIiiESaISB_EERKNS5_IS3_EE[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE12solvePatternERNS_17SparseBlockMatrixINS2_IdLin1ELin1ELi0ELin1ELin1EEEEERKSt6vectorISt4pairIiiESaISB_EERKNS5_IS3_EE]+0xdd): undefined reference to `g2o::csparse_extension::cs_chol_workspace(cs_di_sparse const*, cs_di_symbolic const*, int*, double*)'
main.o: In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solveBlocks(double**&, g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&)':
main.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE11solveBlocksERPPdRKNS_17SparseBlockMatrixIS3_EE[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE11solveBlocksERPPdRKNS_17SparseBlockMatrixIS3_EE]+0xfd): undefined reference to `g2o::csparse_extension::cs_chol_workspace(cs_di_sparse const*, cs_di_symbolic const*, int*, double*)'
main.o: In function `g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solve(g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&, double*, double*)':
main.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0xe8): undefined reference to `g2o::csparse_extension::cs_cholsolsymb(cs_di_sparse const*, double*, cs_di_symbolic const*, double*, int*)'
main.cpp:(.text._ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_[_ZN3g2o19LinearSolverCSparseIN5Eigen6MatrixIdLi6ELi6ELi0ELi6ELi6EEEE5solveERKNS_17SparseBlockMatrixIS3_EEPdS9_]+0x1cc): undefined reference to `g2o::csparse_extension::writeCs2Octave(char const*, cs_di_sparse const*, bool)'
Makefile:239: recipe for target 'pose_estimation_3d2d' failed
collect2: error: ld returned 1 exit status
make: *** [pose_estimation_3d2d] Error 1

zhkmxx9302013, 2018-06-01:
这个问题和ch6的g2o一样(#43)， 是新版的g2o造成的，BlockSover的构造器是unique_ptr的，unique_ptr不支持拷贝构造，所以传参必须通过move，我把改的地方(BA方法的初始化部分)发到下面，可供参考。

``` c++
void bundleAdjustment (
    const vector< Point3f > points_3d,
    const vector< Point2f > points_2d,
    const Mat& K,
    Mat& R, Mat& t )
{
    // 初始化g2o
    // 初始化部分需要改成unique_ptr的， 2018年06月01日17:54:25
    typedef g2o::BlockSolver< g2o::BlockSolverTraits<6,3> > Block;  // pose 维度为 6, landmark 维度为 3
    std::unique_ptr<Block::LinearSolverType> linearSolver (new g2o::LinearSolverCSparse<Block::PoseMatrixType>()); // 线性方程求解器
    std::unique_ptr<Block> solver_ptr( new Block(std::move(linearSolver)));     // 矩阵块求解器
    g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg ( std::move(solver_ptr) );
    g2o::SparseOptimizer optimizer;
    optimizer.setAlgorithm ( solver );

    // vertex(se(3), R, t
    g2o::VertexSE3Expmap* pose = new g2o::VertexSE3Expmap(); // camera pose
    Eigen::Matrix3d R_mat;
    R_mat <<
               R.at<double> ( 0,0 ), R.at<double> ( 0,1 ), R.at<double> ( 0,2 ),
               R.at<double> ( 1,0 ), R.at<double> ( 1,1 ), R.at<double> ( 1,2 ),
               R.at<double> ( 2,0 ), R.at<double> ( 2,1 ), R.at<double> ( 2,2 );

    pose->setId ( 0 );
    pose->setEstimate ( g2o::SE3Quat (  //si yuan shu {R}, t
                            R_mat,
                            Eigen::Vector3d ( t.at<double> ( 0,0 ), t.at<double> ( 1,0 ), t.at<double> ( 2,0 ) )
                        ) 
    );

    optimizer.addVertex ( pose );

    // vertex (Pos in world axes)
    int index = 1;
    for ( const Point3f p:points_3d )   // landmarks
    {
        g2o::VertexSBAPointXYZ* point = new g2o::VertexSBAPointXYZ();
        point->setId ( index++ );
        point->setEstimate ( Eigen::Vector3d ( p.x, p.y, p.z ) );
        point->setMarginalized ( true ); // g2o 中必须设置 marg 参见第十讲内容
        optimizer.addVertex ( point );
    }

    // parameter: camera intrinsics
    g2o::CameraParameters* camera = new g2o::CameraParameters (
        K.at<double> ( 0,0 ), Eigen::Vector2d ( K.at<double> ( 0,2 ), K.at<double> ( 1,2 ) ), 0
    );
    camera->setId ( 0 );
    optimizer.addParameter ( camera );

    // edges
    index = 1;
    for ( const Point2f p:points_2d )
    {
        g2o::EdgeProjectXYZ2UV* edge = new g2o::EdgeProjectXYZ2UV();
        edge->setId ( index );
        edge->setVertex ( 0, dynamic_cast<g2o::VertexSBAPointXYZ*> ( optimizer.vertex ( index ) ) );
        edge->setVertex ( 1, pose );
        edge->setMeasurement ( Eigen::Vector2d ( p.x, p.y ) );
        edge->setParameterId ( 0,0 );
        edge->setInformation ( Eigen::Matrix2d::Identity() );
        optimizer.addEdge ( edge );
        index++;
    }

    chrono::steady_clock::time_point t1 = chrono::steady_clock::now();
    optimizer.setVerbose ( true );
    optimizer.initializeOptimization();
    optimizer.optimize ( 100 );
    chrono::steady_clock::time_point t2 = chrono::steady_clock::now();
    chrono::duration<double> time_used = chrono::duration_cast<chrono::duration<double>> ( t2-t1 );
    cout<<"optimization costs time: "<<time_used.count() <<" seconds."<<endl;

    cout<<endl<<"after optimization:"<<endl;
    cout<<"T="<<endl<<Eigen::Isometry3d ( pose->estimate() ).matrix() <<endl;
}
```

jackxujh, 2019-04-03:
在我的环境中，为了解决这个问题需要重新安装依赖，重新编译和安装g2o，最后再重新编译课本代码。

灵感来源：
- [g2o/csparse_extension link errors · Issue #56 · introlab/rtabmap](https://github.com/introlab/rtabmap/issues/56)
- [Failed to link to g2o on compliation · Issue #2 · jserv/lsd_slam](https://github.com/jserv/lsd_slam/issues/2)