gaoxiang12/slambook #100: 7.3.1节，对极约束推导中的问题

DreamWaterFound, 2018-04-21:
感觉P142页式7.4的写法并不严谨——推导时7.2式使用了齐次坐标意义下的相等，虽然p1、p2还是指那个像素，但是三个坐标的数值还是变了，不应该还使用p1,p2这个符号，容易搞混，如果记为p1~,p2~则p1~=s1p1,p2~=s2p2。如果不考虑使用齐次坐标意义的相等的话，由式7.1有：
[image]
若使x1 x2为两个像素点的归一化平面上的坐标，那么它们满足
[image]
以及满足式7.3：
[image]
那么，将上面的式子代入书本的式7.4中，左右侧分别有：
[image]
[image]
观察上面两个等式的结果，可以发现他们并不相等（而且也不仅仅是只差一个系数）。书本上之所以可以推导出式7.4，就是因为把式7.3中归一化的p1 p2当作是式7.2中的p1 p2了（因为我发现只有这样想，才能够得到式7.4）——即考虑到齐次坐标意义下的相等的时候，式7.3可以由
      x1=K^-1(p1~)
=> Kx1=s1p1
即P86页中式5.7的形式。但是这里p1是像素坐标，式5.7中最后的P是空间点在相机坐标系下的坐标，所以我觉得这样做并不是严谨的，推测这里可能有疏漏。

我个人觉得这部分的推导如果这样写的话会好理解一些，直接截图吧：
由
[image]

稀稀拉拉说了一堆，不知道能不能把问题说明白，希望高博有时间看一下吧。

xiaoerlaigeid, 2018-10-03:
我在看这部份时候也有点晕，我感觉你说的有道理，书里的式子我没有特别清楚，不过看了你的推导清楚了很多。

gaoxiang12, 2018-10-03:
感谢，下一版中会区分正常等号和equal up to scale，你的推导是正确的。

memememeda, 2018-11-24:
的确，谢谢LZ

gaoxiang12, 2018-11-25:
[image]
第二版将修改成如下这个样子，意义和你表达的一致。

DreamWaterFound, 2019-05-25:
灰常感谢高博！😊

closed 2019-05-25 by DreamWaterFound