gaoxiang12/slambook #103: g2o最新版本使用LinearSolverCSparse时出错

chengfzy, 2018-05-03:
下载编译最新的g2o版本(至少今年的版本)，如果使用LinearSolverCSparse就会出错，比如/ch6/g2o_curve_fitting的代码，原始是使用的LinearSolverDense：
```
    // construct graph optimization
    using Block = BlockSolver<g2o::BlockSolverTraits<3, 1>>;  // Block, parameters dims = 3, residual dim = 1
    // linear solver, dense incremental equation
    unique_ptr<Block::LinearSolverType> linearSolver{new LinearSolverDense<Block::PoseMatrixType>()};
    unique_ptr<Block> solver{new Block(move(linearSolver))};  // block solver
    // graph solver, could be Gaussian-Newton, LM, and DogLeg
    OptimizationAlgorithmLevenberg* optAlg = new OptimizationAlgorithmLevenberg(move(solver));
    SparseOptimizer optimizer;  // graph model
    optimizer.setAlgorithm(optAlg);
    optimizer.setVerbose(true);  // debug output
```
将LinearSolverDense改成LinearSolverCSparse后:
`unique_ptr<Block::LinearSolverType> linearSolver{new LinearSolverCSparse<Block::PoseMatrixType>()};`
出现错误：
/usr/bin/ld: CMakeFiles/Eg06_UseG2O.dir/main.cpp.o: undefined reference to symbol 'cs_di_calloc'
//usr/lib/x86_64-linux-gnu/libcxsparse.so.3.1.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [Eg06_UseG2O/Eg06_UseG2O] Error 1
make[1]: *** [Eg06_UseG2O/CMakeFiles/Eg06_UseG2O.dir/all] Error 2
make: *** [all] Error 2

其他相关的g2o代码一样会出现问题，但g2o自带的sba example却是可以运行的，google了也没有找到原因，试过重新安装新虚拟机，同样问题。

环境：ubuntu14.04, 已经安装suitsparse, csparse并cmake链接相关库

chengfzy, 2018-05-24:
Update:问题没有解决，不清楚原因，目前使用git ff647bd7版本

mc275, 2018-07-05:
我在加了以下内容以后搞定了，貌似应该是缺少了*cxsparse*
```
SET(G2O_LIBS g2o_cli g2o_ext_freeglut_minimal g2o_simulator g2o_solver_slam2d_linear g2o_types_icp g2o_types_slam2d g2o_core g2o_interface g2o_solver_csparse g2o_solver_structure_only g2o_types_sba g2o_types_slam3d g2o_csparse_extension g2o_opengl_helper g2o_solver_dense g2o_stuff g2o_types_sclam2d g2o_parser g2o_solver_pcg g2o_types_data g2o_types_sim3 cxsparse )

target_link_libraries(g2o_test ${G2O_LIBS})

gaoxiang12, 2018-07-16:
closed because the problem is solved.

closed 2018-07-16 by gaoxiang12

eavae, 2020-07-24:
这里留个我的解决思路吧，问题的本质是最新版的g2o为了更安全的内存，强制在传参的时候需要unique指针。方法如下：

`// 线性方程求解器
    auto linear_solver = g2o::make_unique<g2o::LinearSolverCSparse<g2o::BlockSolver_6_3::PoseMatrixType>>();
    // 矩阵块求解器
    auto block_solver = g2o::make_unique<g2o::BlockSolver_6_3>(std::move(linear_solver));
    g2o::OptimizationAlgorithmLevenberg *solver = new g2o::OptimizationAlgorithmLevenberg(std::move(block_solver));
    g2o::SparseOptimizer optimizer;
    optimizer.setAlgorithm(solver);`

X-NEARS, 2023-02-14:
感谢[MC275](https://github.com/mc275)提供的建议
> SET(G2O_LIBS g2o_cli g2o_ext_freeglut_minimal g2o_simulator g2o_solver_slam2d_linear g2o_types_icp g2o_types_slam2d g2o_core g2o_interface g2o_solver_csparse g2o_solver_structure_only g2o_types_sba [...]
target_link_libraries(g2o_test ${G2O_LIBS})

如果在添加之后仍然出现
>fatal error: cs.h: 没有那个文件或目录
 #include <cs.h>

那么可以通过以下，定位cs.h文件位置
>dpkg -L libsuitesparse-dev | grep 'cs.h'

之后添加至cmakelist
>include_directories("/usr/include/suitesparse") 
SET(G2O_LIBS g2o_cli g2o_ext_freeglut_minimal g2o_simulator g2o_solver_slam2d_linear g2o_types_icp g2o_types_slam2d g2o_core g2o_interface g2o_solver_csparse g2o_solver_structure_only g2o_types_sba g2o_types_slam3d g2o_csparse_extension g2o_opengl_helper g2o_solver_dense g2o_stuff g2o_types_sclam2d g2o_parser g2o_solver_pcg g2o_types_data g2o_types_sim3 cxsparse )
target_link_libraries(g2o_test ${G2O_LIBS})

注意修改“g2o_test”