gaoxiang12/slambook #105: Process finished with exit code 134 (interrupted by signal 6: SIGABRT)

XZLeo, 2018-05-10:
高博士您好：
我之前遇到了Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)这样的问题，就去Github上复制粘贴了一遍代码，运行之后出现了新的问题，运行后像这样：

正在将图像转换为点云...
转换图像中: 1
转换图像中: 2
转换图像中: 3
转换图像中: 4
转换图像中: 5
点云共有0个点.
terminate called after throwing an instance of 'pcl::IOException'
  what():  : [pcl::PCDWriter::writeBinary] Input point cloud has no data!

Process finished with exit code 134 (interrupted by signal 6: SIGABRT)

刚入门c++，很多不懂之处，经常跑不出例程，希望您的帮助，谢谢！

gs14iitbbs, 2018-07-02:
Use this always 

if(pointCloud->size() > 0)
        pcl::io::savePCDFileBinary("map.pcd", *pointCloud );

Though I still need to know, why  点云共有0个点. ?

Haotian-Zhang, 2018-07-16:
你好，
报错的原因是因为你修改之后pose.txt找到了，但是color和depth文件你还是得加上../color和../depth，或者就是将编译joinMap的文件放置在pose.txt目录下执行

closed 2020-07-25 by XZLeo