gaoxiang12/slambook #108: ch5 imageBasic make报错

TerenceCYJ, 2018-05-18:
cyj@dell-cyj:~/github_project/slambook-master/ch5/imageBasics/build$ make
[ 50%] Linking CXX executable imageBasics
/usr/bin/ld: warning: libjpeg.so.9, needed by //home/cyj/miniconda3/lib/libtiff.so.5, not found (try using -rpath or -rpath-link)
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFReadRGBAStrip@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFSetErrorHandler@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFReadEncodedStrip@LIBTIFF_4.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_std_error@LIBJPEG_9.0'
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFReadRGBATile@LIBTIFF_4.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_read_header@LIBJPEG_9.0'
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFClientOpen@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFSetField@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFScanlineSize@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFSetWarningHandler@LIBTIFF_4.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_suppress_tables@LIBJPEG_9.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_CreateDecompress@LIBJPEG_9.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_finish_decompress@LIBJPEG_9.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_CreateCompress@LIBJPEG_9.0'
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFWriteScanline@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFRGBAImageOK@LIBTIFF_4.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_write_tables@LIBJPEG_9.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_resync_to_restart@LIBJPEG_9.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_set_defaults@LIBJPEG_9.0'
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFReadDirectory@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFIsTiled@LIBTIFF_4.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_abort@LIBJPEG_9.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_start_decompress@LIBJPEG_9.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_read_scanlines@LIBJPEG_9.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_start_compress@LIBJPEG_9.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_destroy@LIBJPEG_9.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_write_raw_data@LIBJPEG_9.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_set_quality@LIBJPEG_9.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_finish_compress@LIBJPEG_9.0'
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFReadEncodedTile@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFClose@LIBTIFF_4.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_read_raw_data@LIBJPEG_9.0'
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFNumberOfStrips@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFWriteEncodedStrip@LIBTIFF_4.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_has_multiple_scans@LIBJPEG_9.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_set_colorspace@LIBJPEG_9.0'
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFGetField@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.3.3.1: undefined reference to `TIFFOpen@LIBTIFF_4.0'
//home/cyj/miniconda3/lib/libtiff.so.5: undefined reference to `jpeg_write_scanlines@LIBJPEG_9.0'
collect2: error: ld returned 1 exit status
CMakeFiles/imageBasics.dir/build.make:151: recipe for target 'imageBasics' failed
make[2]: *** [imageBasics] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/imageBasics.dir/all' failed
make[1]: *** [CMakeFiles/imageBasics.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

TerenceCYJ, 2018-05-31:
Something wrong with opencv

closed 2018-05-31 by TerenceCYJ

yuyangyg, 2018-06-14:
hi,I had the same problem. have you corrected it?

yuyangyg, 2018-06-14:
I have corrected it ,by add    target_link_libraries( imageBasics "/home/yuyangyg/anaconda2/lib/libjpeg.so.9" )