gaoxiang12/slambook #111: 我在运行第九章的project工程时，出现了double free的错误？

lugui2009, 2018-06-06:
我在运行第九章的project工程时，0.1和0.2都顺利通过，但是在运行0.3和0.4时都是报错：
double free or corruption (out)
Aborted (core dumped)

我的ubuntu系统是18.04LTS ,有人出现过这样的错误吗？求指导。

jucic, 2018-07-16:
我跑第七章pnp和icp的时候就报double free or corruption了

LevinJ, 2019-02-22:
I am using Ubuntu 18.04, and encountered the same issue.  I have it fixed by removing the `-march=native` from the CMAKE_CXX_FLAGS flag in CMakeList.txt file.

I figured out this fix by reading this link, https://github.com/introlab/rtabmap/issues/343

gaochuan2017, 2019-03-21:
> I am using Ubuntu 18.04, and encountered the same issue. I have it fixed by removing the `-march=native` from the CMAKE_CXX_FLAGS flag in CMakeList.txt file.
> 
> I figured out this fix by reading this link, [introlab/rtabmap#343](https://github.com/introlab/rtabmap/issues/343)

it really works bro,thanks a lot

John1102, 2019-04-06:
> I am using Ubuntu 18.04, and encountered the same issue. I have it fixed by removing the `-march=native` from the CMAKE_CXX_FLAGS flag in CMakeList.txt file.
> 
> I figured out this fix by reading this link, [introlab/rtabmap#343](https://github.com/introlab/rtabmap/issues/343)

Thank you so much!!!!

fuhuaida, 2019-08-04:
你的这个问题解决了吗？我的也是ubun18.04运行project0.3 0.4的时候出现这种错误，可否告知一下呢？qq： 460593411，感谢

caiyongqi, 2019-10-20:
我的系统是unbutu18.04,在第七章BA优化时代码报错double free or corruption (out)，找到这儿来后发现CMakeList.txt中并没有march=native，但是加上后可以正常运行。[并不知道原因，花了很长时间才解决，所以在这记录一下，希望能帮到有类似情况的人]

LongruiDong, 2020-10-15:
我在ubunut18.04跑ORBSLAM时也遇到这个问题，去掉ORBSLAM Cmakelists.txt 中的march=native，运行就没corrupt了，我怀疑和eigen版本也有关，我从3.2.10升级到3.3.2后遇到这问题，