gaoxiang12/slambook #119: project 0.2 编译visual_odometry.cpp报错

HanochZzhou, 2018-06-28:
我在编译0.2版的工程时出现以下错误，还麻烦各位大神解答一下：
[ 66%] Building CXX object src/CMakeFiles/myslam.dir/visual_odometry.cpp.o
[ 77%] Linking CXX shared library ../../lib/libmyslam.so
[ 77%] Built target myslam
Scanning dependencies of target run_vo
[ 88%] Building CXX object test/CMakeFiles/run_vo.dir/run_vo.cpp.o
[100%] Linking CXX executable ../../bin/run_vo
CMakeFiles/run_vo.dir/run_vo.cpp.o: In function `main':
run_vo.cpp:(.text.startup+0x95c): undefined reference to `cv::viz::Viz3d::Viz3d(cv::String const&)'
run_vo.cpp:(.text.startup+0x97c): undefined reference to `cv::viz::WCoordinateSystem::WCoordinateSystem(double)'
run_vo.cpp:(.text.startup+0x990): undefined reference to `cv::viz::WCoordinateSystem::WCoordinateSystem(double)'
run_vo.cpp:(.text.startup+0xa1d): undefined reference to `cv::viz::makeCameraPose(cv::Vec<double, 3> const&, cv::Vec<double, 3> const&, cv::Vec<double, 3> const&)'
run_vo.cpp:(.text.startup+0xa2c): undefined reference to `cv::viz::Viz3d::setViewerPose(cv::Affine3<double> const&)'
run_vo.cpp:(.text.startup+0xa45): undefined reference to `cv::viz::Widget::setRenderingProperty(int, double)'
run_vo.cpp:(.text.startup+0xa5e): undefined reference to `cv::viz::Widget::setRenderingProperty(int, double)'
run_vo.cpp:(.text.startup+0xad5): undefined reference to `cv::viz::Viz3d::showWidget(cv::String const&, cv::viz::Widget const&, cv::Affine3<double> const&)'
run_vo.cpp:(.text.startup+0xb7a): undefined reference to `cv::viz::Viz3d::showWidget(cv::String const&, cv::viz::Widget const&, cv::Affine3<double> const&)'
run_vo.cpp:(.text.startup+0x1392): undefined reference to `cv::viz::Viz3d::setWidgetPose(cv::String const&, cv::Affine3<double> const&)'
run_vo.cpp:(.text.startup+0x13b1): undefined reference to `cv::viz::Viz3d::spinOnce(int, bool)'
run_vo.cpp:(.text.startup+0x1534): undefined reference to `cv::viz::Widget::~Widget()'
run_vo.cpp:(.text.startup+0x1540): undefined reference to `cv::viz::Widget::~Widget()'
run_vo.cpp:(.text.startup+0x154c): undefined reference to `cv::viz::Viz3d::~Viz3d()'
run_vo.cpp:(.text.startup+0x1dc3): undefined reference to `cv::viz::Widget::~Widget()'
run_vo.cpp:(.text.startup+0x1dcf): undefined reference to `cv::viz::Widget::~Widget()'
run_vo.cpp:(.text.startup+0x1ddb): undefined reference to `cv::viz::Viz3d::~Viz3d()'
collect2: error: ld returned 1 exit status
test/CMakeFiles/run_vo.dir/build.make:124: recipe for target '../bin/run_vo' failed
make[2]: *** [../bin/run_vo] Error 1
CMakeFiles/Makefile2:140: recipe for target 'test/CMakeFiles/run_vo.dir/all' failed
make[1]: *** [test/CMakeFiles/run_vo.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
*** Failure: Exit code 2 ***
谢谢。

gaoxiang12, 2018-07-10:
你的OpenCV可能没有编译viz模块，这在默认的编译配置中是不编译或者可选的，请检查你是否编译并安装了此模块。

HanochZzhou, 2018-07-17:
谢谢您的指导，我把Opencv和vtk全部都重新编译了一遍 最后成功了！谢谢

closed 2018-07-17 by HanochZzhou