gaoxiang12/slambook #121: 第七章3d2d运行时报错（用的新版的g2o但是都改成了unique_ptr所以不是这个指针的问题）

jucic, 2018-07-15:
./pose_estimation_3d2d 1.png 2.png 1_depth.png 2_depth.png
*** Error in `./pose_estimation_3d2d': double free or corruption (out): 0x0000000000d931e0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7fc37d73d7e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7fc37d74637a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7fc37d74a53c]
/usr/local/lib/libg2o_core.so(_ZN3g2o7Factory12registerTypeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEPNS_32AbstractHyperGraphElementCreatorE+0x497)[0x7fc3805e3667]

gaoxiang12, 2018-07-16:
查一下是否是链接的问题，你的机器上是不是安装了两个不同版本的g2o？

jucic, 2018-07-16:
只装了一个版本的g2o，网上查这错误说是指针什么的内存被重复释放？检查了代码好像没这种问题，摸不着头脑。。

jucic, 2018-07-16:
只是运行./pose_estimation_3d2d命令行参数不给全也是报这个错误，而不是提示"usage:./pose_estimation_3d2d img1 img2 depth1 depth2"的错误，很奇怪了。

jackxujh, 2019-04-09:
我在 project/0.3 也遇到这个问题了

HavendGithub, 2019-07-25:
我也是这个问题
*** Error in `/home/haven/slam/slambook/project/0.3/bin/run_vo': double free or corruption (out): 0x0000000002486a20 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f4a7e7577e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f4a7e76037a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f4a7e76453c]
/usr/local/lib/libg2o_types_sba.so(_ZN3g2o26EdgeProjectP2MC_IntrinsicsD0Ev+0x2d)[0x7f4a7e02bc2d]
/usr/local/lib/libg2o_core.so(_ZN3g2o7Factory12registerTypeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEPNS_32AbstractHyperGraphElementCreatorE+0x497)[0x7f4a7e4bdf97]
/usr/local/lib/libg2o_types_sba.so(+0x1bbda)[0x7f4a7e023bda]
/lib64/ld-linux-x86-64.so.2(+0x106ca)[0x7f4a816186ca]
/lib64/ld-linux-x86-64.so.2(+0x107db)[0x7f4a816187db]
/lib64/ld-linux-x86-64.so.2(+0xc6a)[0x7f4a81608c6a]
======= Memory map: ========
00400000-00407000 r-xp 00000000 08:07 12714321                           /home/haven/slam/slambook/project/0.3/bin/run_vo
00606000-00607000 r--p 00006000 08:07 12714321                           /home/haven/slam/slambook/project/0.3/bin/run_vo
00607000-00608000 rw-p 00007000 08:07 12714321                           /home/haven/slam/slambook/project/0.3/bin/run_vo
02420000-024a4000 rw-p 00000000 00:00 0                                  [heap]
7f4a50000000-7f4a50021000 rw-p 00000000 00:00 0 
7f4a50021000-7f4a54000000 ---p 00000000 00:00 0

yxliwhu, 2020-04-14:
将set( CMAKE_CXX_FLAGS "-std=c++11 -march=native -O3" )改成set( CMAKE_CXX_FLAGS "-std=c++11 -O3" )