gaoxiang12/slambook #125: ch7/CMakeList.txt g2o相关

Jinxz666, 2018-07-31:
By refering to game-of-drones's issue of  #ch6/g2o_curve_fitting/CMakeLists.txt, the part of pose_estimation_3d2d & pose_estimation_3d3d should also be adjusted as following to avoid the error 

> while loading shared libraries: libg2o_core.so: cannot open shared object file: No such file or directory

`target_link_libraries(……
   ${G2O_CORE_LIBRARY} ${G2O_STUFF_LIBRARY}
   g2o_types_sba g2o_csparse_extension`

After modifying, I can execute the code successfully. While I still wondering whether should I operate to the last two g2o libraries and what is the difference between ${G2O_CORE_LIBRARY} and g2o_core in target_link libraries.

gaoxiang12, 2018-08-01:
Take a look at FindG2O.cmake file which defines a macro to find g2o library and set their names. Almost all g2o_types_xxx are defined here.
The FindG2O.cmake is in cmake_module directory of many projects that rely on g2o. For example here: https://github.com/gaoxiang12/slambook/blob/master/ch6/g2o_curve_fitting/cmake_modules/FindG2O.cmake

huangwen0907, 2019-08-30:
> sudo gedit /etc/ld.so.conf

添加如下代码：
> /usr/local/lib

运行：
> sudo ldconfig

hanzheteng, 2020-03-25:
According to [this answer](https://unix.stackexchange.com/a/22999), the following way also works:
```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib
```
This is to temporarily add the path `/usr/local/lib` to the shared library search path. You can run this command before running g2o program. 

It is also possible to add this line to the end of `~/.bashrc` file, such that you don't need to run this command before running g2o program everytime.