gaoxiang12/slambook #127: 学长，按照课本cmake安装好g2o库之后，编译您github源码第七章的时候出现以下问题

w8128835, 2018-08-07:
学长您好，打扰您了。
我的g2o_core等文件在'/home/eric/g2o/lib/libg2o_core.so'这个位置，但是make您git里面第七章源码的时候出现下面报错：
[ 40%] Built target pose_estimation_2d2d
Linking CXX executable pose_estimation_3d2d
/usr/bin/ld: 找不到 -lg2o_core
/usr/bin/ld: 找不到 -lg2o_stuff
/usr/bin/ld: 找不到 -lg2o_types_sba
/usr/bin/ld: 找不到 -lg2o_csparse_extension
collect2: error: ld returned 1 exit status
make[2]: *** [pose_estimation_3d2d] 错误 1
请问这个怎么解决

w8128835, 2018-08-08:
确实是这样，非常感谢学长^_^

closed 2018-08-08 by gaoxiang12

shaoxh, 2021-02-14:
你好，我 make install 了，但是还有这个的报错。我查找了 在 /usr/local/lib/ 下面都有这些文件的。
-rw-r--r--  1 root root    766192 2月  14 15:59 libg2o_cli_d.so
-rw-r--r--  1 root root  20440944 2月  14 15:59 libg2o_core_d.so
-rw-r--r--  1 root root    225304 2月  14 15:59 libg2o_csparse_extension_d.so
-rw-r--r--  1 root root    178816 2月  14 15:59 libg2o_ext_csparse_d.so
... ....
请问你知道是什么原因吗？ @w8128835 @gaoxiang12

shaoxh, 2021-02-19:
通过以下命令重新删除 g2o 安装后解决了。
应该是之前版本的没有删除干净。
我在网上看到其他删除 g2o 的方法删不干净。
供大家参考
$ sudo rm -r /usr/local/include/EXTERNAL /usr/local/include/g2o
$ sudo rm -r /usr/local/lib/libg2o_*
$ cd g2o/build
$ rm -rf *
$ cmake ..
$ make -j4
$ sudo make install