gaoxiang12/slambook #129: CH5 joinMap error segmentation fault (core dumped)

YuYuCong, 2018-08-20:
# 问题描述
编译已成功，并且是在pose.txt所在文件夹下面运行./joinMap，结果报错segmentation fault (core dumped)，请教一下是否有解决方案，或者潜在的解决办法？

gaoxiang12, 2018-08-20:
* 检查图像是否正确读取
* 如果其他地方都对，把vector的第二个模板参数设成Eigen::aligned_allocator<>。不用内存对齐的allocator在某些机器上可能导致问题。

YuYuCong, 2018-08-20:
#### 问题分析
感谢高博的帮助，问题已得到解决。
我逐行检查了代码，最终发现是在最后保存点云图时`pcl::io::savePCDFileBinary("map.pcd", *pointCloud );`这一句代码导致出错。
于是尝试重新安装PCL点云库的时候发现依赖libvtk5-qt4-dev已经被破坏，重新安装了libvtk5-qt4-dev依赖，点云图正常保存，bug解决。

## 总结如下：
#### 问题描述
  安装PCL依赖时libvtk5-qt4-dev无法安装，
  报错信息：E: Unable to correct problems, you have held broken packages.

  $ sudo apt-get install libvtk5-qt4-dev
  Reading package lists...
  Building dependency tree...
  Reading state information...
  Some packages could not be installed. This may mean that you have
  requested an impossible situation or if you are using the unstable
  distribution that some required packages have not yet been created
  or been moved out of Incoming.
  The following information may help to resolve the situation:

  The following packages have unmet dependencies:
   libvtk5-qt4-dev : Depends: libvtk5-dev (= 5.10.1+dfsg-2.1build1) but it is not going to be installed
  E: Unable to correct problems, you have held broken packages.
  #### 问题分析
  应该是当初安装完PCL，后来安装其他东西做了一些改动，初步推断是ROS依赖
#### 解决方案
  $ sudo apt-get upgrade
  $ sudo apt-get update
先安装
   $ sudo apt-get install libvtk5-dev
   然后再安装
  $ sudo apt-get install libvtk5-qt4-dev
#### 重要记录
   为了解决上述问题，似乎我的ROS依赖被删掉了不少
  【警告】是否会影响到ROS？未知！如果有小伙伴在使用PCL的同时使用ros可能会遇到这个问题
  ===================================solved======================================

YuYuCong, 2019-01-28:
> 楼主的pcl是在ubuntu 16.04上装的么？

是的，安装记录：[https://github.com/HustRobot/VSLAM](https://github.com/HustRobot/VSLAM)