gaoxiang12/slambook #141: ch4编译时出现问题

161180134, 2018-10-05:
CMakeFiles/useSophus.dir/useSophus.cpp.o：在函数‘main’中：
useSophus.cpp:(.text+0xd2)：对‘Sophus::SO3::SO3(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&)’未定义的引用
useSophus.cpp:(.text+0xf5)：对‘Sophus::SO3::SO3(double, double, double)’未定义的引用
useSophus.cpp:(.text+0x127)：对‘Sophus::SO3::SO3(Eigen::Quaternion<double, 0> const&)’未定义的引用
useSophus.cpp:(.text+0x1f7)：对‘Sophus::SO3::log() const’未定义的引用
useSophus.cpp:(.text+0x279)：对‘Sophus::SO3::hat(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x2cf)：对‘Sophus::SO3::hat(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x2e8)：对‘Sophus::SO3::vee(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&)’未定义的引用
useSophus.cpp:(.text+0x38a)：对‘Sophus::SO3::exp(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x3aa)：对‘Sophus::SO3::operator*(Sophus::SO3 const&) const’未定义的引用
useSophus.cpp:(.text+0x47b)：对‘Sophus::SE3::SE3(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x49b)：对‘Sophus::SE3::SE3(Eigen::Quaternion<double, 0> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x558)：对‘Sophus::SE3::log() const’未定义的引用
useSophus.cpp:(.text+0x5ef)：对‘Sophus::SE3::hat(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x645)：对‘Sophus::SE3::hat(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x65e)：对‘Sophus::SE3::vee(Eigen::Matrix<double, 4, 4, 0, 4, 4> const&)’未定义的引用
useSophus.cpp:(.text+0x6f7)：对‘Sophus::SE3::exp(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)’未定义的引用
useSophus.cpp:(.text+0x717)：对‘Sophus::SE3::operator*(Sophus::SE3 const&) const’未定义的引用
useSophus.cpp:(.text+0x75b)：对‘Sophus::SE3::matrix() const’未定义的引用
CMakeFiles/useSophus.dir/useSophus.cpp.o：在函数‘Sophus::operator<<(std::ostream&, Sophus::SO3 const&)’中：
useSophus.cpp:(.text._ZN6SophuslsERSoRKNS_3SO3E[_ZN6SophuslsERSoRKNS_3SO3E]+0x2e)：对‘Sophus::SO3::log() const’未定义的引用
collect2: error: ld returned 1 exit status
CMakeFiles/useSophus.dir/build.make:94: recipe for target 'useSophus' failed
make[2]: *** [useSophus] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/useSophus.dir/all' failed
make[1]: *** [CMakeFiles/useSophus.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

gaoxiang12, 2018-10-06:
貌似没链接sohpus库，检查你的cmakelists

161180134, 2018-10-06:
> 貌似没链接sohpus库，检查你的cmakelists

看了一下CmakeList.txt,感觉没问题呀。就是下面几行
cmake_minimum_required( VERSION 2.8 )
project( useSophus )

# 为使用 sophus，您需要使用find_package命令找到它
find_package( Sophus REQUIRED )
include_directories( ${Sophus_INCLUDE_DIRS} )

add_executable( useSophus useSophus.cpp )
target_link_libraries( useSophus ${Sophus_LIBRARIES} )

deeprun, 2021-03-14:
I have the same problem. How did  you fix it?