gaoxiang12/slambook #143: ch4 编译  useSophus 报错

WithFoxSquirrel, 2018-10-18:
编译useSophus 报错

  /usr/local/include/sophus/common.hpp:44:10: error: 'auto' not allowed in function return type
    static auto test(int)
           ^~~~
  /usr/local/include/sophus/common.hpp:44:24: error: expected ';' at end of declaration list
    static auto test(int)

##  查资料有说是c++11的问题，cmakelists增加 set( CMAKE_CXX_FLAGS "-std=c++14 -O3" ) 后，编译报

   /Users/hans.hu/work/slam/slambook/ch4/useSophus/useSophus.cpp:16:13: error: use of class template 'Sophus::SO3' requires template arguments
       Sophus::SO3 SO3_R(R);               // Sophus::SO(3)可以直接从旋转矩阵构造
               ^
   /usr/local/include/sophus/so3.hpp:423:7: note: template is declared here
   class SO3 : public SO3Base<SO3<Scalar_, Options>> {

gaoxiang12, 2018-10-19:
书里代码用的是早期非模板版本的sophus。在git clone sophus之后需要checkout到指定版本才能编译。

roachsinai, 2018-11-29:
不知高博能否告知，你所使用的gcc版本？

LvjianLu, 2022-07-20:
> [image]
> 
> > 按照书上，checkout到 git checkout a621ff (实际是a621ff2)
> 
> [image]
> 
> > 编译出错，修改unit_complex_ real() = 1.0 ;   =>   unit_complex_.real(1.0); 编译成功
> 
> > 然后到ch4/useSophus/build 目录， make
> 
> [image]
> 
> > clang 版本和平台 Apple LLVM version 9.1.0 (clang-902.0.39.2) , update_se3(0,0) = 1e-4d; -> update_se3(0,0) = 1e-4;
> 
> [image]
> 
> > 高博，书上的有些例子不能在 osx 上编译吗？

分别改为：
unit_complex_.real(1.);
unit_complex_.imag(0.);
以及
update_se3(0,0) = 1e-4;
就可以编译通过

closed 2023-07-20 by WithFoxSquirrel

BTREE-C802, 2023-07-20:
您好，我叫旷鸿章，我已经收到了你的邮件。谢谢！