gaoxiang12/slambook #147: CH5 安装pcl库，执行apt-get update后报错。。。这个怎么解决？

happyrookie, 2018-11-09:
W: 仓库 “http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu xenial Release” 没有 Release 文件。
N: 无法认证来自该源的数据，所以使用它会带来潜在风险。
N: 参见 apt-secure(8) 手册以了解仓库创建和用户配置方面的细节。
E: 无法下载 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu/dists/xenial/main/binary-amd64/Packages  404  Not Found
E: 部分索引文件下载失败。如果忽略它们，那将转而使用旧的索引文件。

然后我不管这个错误直接执行apt-get install libpcl-all就出现了下面这个错误：
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
E: 无法定位软件包 libpcl-all

我是菜鸟，还请大家指导一下。。。

gaoxiang12, 2018-11-09:
网络原因吧，如果一直连不上就先跳过这个实验好了。

happyrookie, 2018-11-09:
我用源代码编译（应该是源代码）又告诉我FLANN的版本要大于1.7，这个FLANN不是OPENCV里面的一个函数嘛?我下载的OPENCV已经是4.0以下最新的了啊。。。

happyrookie, 2018-11-09:
> 网络原因吧，如果一直连不上就先跳过这个实验好了。

我用源代码编译（应该是源代码）又告诉我FLANN的版本要大于1.7，这个FLANN不是OPENCV里面的一个函数嘛?我下载的OPENCV已经是4.0以下最新的了啊。。。

happyrookie, 2018-11-09:
费了九牛二虎之力终于安上了。。

gaoxiang12, 2018-11-15:
源码编译可能会出一些问题，下一版书我不准备用pcl了，可视化统一用pangolin。

happyrookie, 2018-11-15:
对linux只会基本的操作，所以出现了许多问题幸亏有csdn

ran5515, 2018-11-22:
应该不是网络问题。我也遇到了这个错误，请问@happyrookie 你是怎么解决的？谢谢。

ran5515, 2018-11-22:
我是指“E: 无法下载 http://ppa.launchpad.net/v-launchpad-jochen-sprickerhof-de/pcl/ubuntu/dists/xenial/main/binary-amd64/Packages 404 Not Found
E: 部分索引文件下载失败。如果忽略它们，那将转而使用旧的索引文件。”这个问题
谢谢

AnkyZhang, 2020-01-22:
> > 网络原因吧，如果一直连不上就先跳过这个实验好了。
> 
> 我用源代码编译（应该是源代码）又告诉我FLANN的版本要大于1.7，这个FLANN不是OPENCV里面的一个函数嘛?我下载的OPENCV已经是4.0以下最新的了啊。。。
> CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
> Could NOT find FLANN (missing: FLANN_LIBRARY FLANN_INCLUDE_DIR) (Required
> is at least version "1.7.0")
> Call Stack (most recent call first):
> /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
> cmake/Modules/FindFLANN.cmake:54 (find_package_handle_standard_args)
> CMakeLists.txt:249 (find_package)

请问这个问题怎么解决呀？