gaoxiang12/slambook #154: ch6 g2o_curve_fitting中的代码由于g2o的更新而编译不过，另外用G-N方法来优化的时候效果较差

rokid-chrislyu, 2018-11-27:
我在跑ch6中的g2o_curve_fitting的代码的时候：

1. 如果从github上clone最新的g2o, 那么书中的样例代码中第76行和77行中的指针要改成unique_ptr, 并且79行改成接受unique_ptr的参数后才能编译通过，g2o做了比较大的重构？

2. 该代码中，我如果跑L-M method或者dogleg method, 结果相符，但是如果跑G-N method，结果差得很大，不知道是什么原因？

gaoxiang12, 2018-11-28:
1. g2o在17年左右把原来的祼指针统一改成了智能指针。代码的写法有少许改变，你可以看g2o自带的examples。
2. g2o的G-N实现的不太好，通常用g2o都是L-M或dogleg。

chenchen2015, 2019-01-13:
以下对于有同样问题的人可能有用：

[g2o repo的这个issue](https://github.com/RainerKuemmerle/g2o/issues/290#issuecomment-410230679)以及[g2o官方示例源码](https://github.com/RainerKuemmerle/g2o/blob/master/g2o/examples/data_fitting/curve_fit.cpp#L156)有关于新的智能指针写法的实例，需要注意的是g2o有自己的`make_unique`实现`g2o::make_unique`，实际上没必要用`std::make_unique`（否则还需要修改`CMakeLists.txt`支持到`c++14`标准，并同时包含`<memory>`头文件），不管用哪个结果是一样的

另外 @gaoxiang12 因为g2o应该会保持使用智能指针的实现方式了，选择从GitHub直接clone最新版本库的人应该都会遇见同样问题，有没有计划更新现有代码来适应g2o？

gaoxiang12, 2019-01-14:
第二版会统一用智能指针的版本。

> 以下对于有同样问题的人可能有用：
> 
> [g2o repo的这个issue](https://github.com/RainerKuemmerle/g2o/issues/290#issuecomment-410230679)以及[g2o官方示例源码](https://github.com/RainerKuemmerle/g2o/blob/master/g2o/examples/data_fitting/curve_fit.cpp#L156)有关于新的智能指针写法的实例，需要注意的是g2o有自己的`make_unique`实现`g2o::make_unique`，实际上没必要用`std::make_unique`（否则还需要修改`CMakeLists.txt`支持到`c++14`标准，并同时包含`<memory>`头文件），不管用哪个结果是一样的
> 
> 另外 @gaoxiang12 因为g2o应该会保持使用智能指针的实现方式了，选择从GitHub直接clone最新版本库的人应该都会遇见同样问题，有没有计划更新现有代码来适应g2o？
> 抱歉，看到已经有pull request了， #110