gaoxiang12/slambook #162: 关于第七章相机运动（外参）的疑问

memememeda, 2018-12-04:
书中7.2节最后写道：

> 接下来，我们希望根据匹配的点对，估计**相机的运动**。这里由于相机的原理不同，情况发生了变化：
> 1. ……
> 2. ……
> 3. 如果我们有3D 点和它们在相机的投影位置，也能估计相机的运动。该问题通过PnP求解。

7.7节，书中写的是： 

> 已知3D点的空间位置和相机上的投影点，求相机的旋转和平移（外参）。

2D-2D的对极几何 和 3D-3D的ICP 求解的都是两帧之间的匹配点的运动，也就是相机的运动；
但是PnP求解的T 是同一帧的3D点到2D点的变换吧，也就是Tcw，这个Tcw不能理解为相机的运动吧？

Zkjoker, 2018-12-21:
你发现的确实比较敏锐，我看了你的疑惑也发现了这个问题。
这部分高博写的好像有点让人迷惑，我的理解是这样的：
你所说的确实是没错的。算出来的确实是Tcw。但是这时候有一个问题，你之前肯定没有考虑过：什么是世界坐标系？凭空怎么安排世界坐标系呢？难道按照上北下南左西右东？肯定不可能。
世界坐标系也是人为指定的，它是一个逻辑上的概念，你可以把相机放到某一位置朝向某一角度，我们暂时把这个位兹记做A。指定这个时刻的A相机坐标系就是世界坐标系。这时相机就当作世界坐标系的原点。
那么你算出的TCW，确实是相机坐标系与世界坐标系的变换，但同样也是从初始位置A到求解出的B位置下相机坐标系的变换，所以它也是相机运动！

高博在书159页下方加粗的字体，写明ABC是世界坐标系下的坐标。这个世界坐标系是像我上面说的，是人为指定的。你如果还有疑虑，你可以看下165页那段代码。第15行，将相机在1.png位置时刻的像素坐标转化为1.png时刻的归一化平面的相机坐标（这个时刻还不知深度），然后在第16行乘以深度，变成实际的1.png位置下的真正的相机坐标，**然后这就作为世界坐标系**，再提供2.png下的2d点位置。恢复出2.png位置下的R和t！那这个R和t确实是Tcw，但也是从1.png到2.png之间的旋转和平移！

所以总结一下，
所谓pnp的目的，就是：
现在有世界坐标系（世界坐标系也就是物体在A位置的相机坐标系下的3d点位置）；
然后有物体在位置B的2D点。
根据上述信息，恢复物体从初始的A时刻到B时刻的相机运动（实际上在B位置观测到的相机坐标系下的3D点也同时知道了：看书161页，根据7.33方程解出x和y，把x和y代入7.32的第一个式子，把v解出来，v知道以后由于AB平方也是知道的，可以得知OC平方，那么深度也就解出来了）

如果有说的不对和不准确的地方，感谢指正！

Zkjoker, 2018-12-21:
更正一下：“现在有世界坐标系（世界坐标系也就是物体在A位置的相机坐标系下的3d点位置）”说错了，应为：“现在有世界坐标系（世界坐标系也就是物体在A位兹的相机坐标系）”

memememeda, 2019-03-06:
> 更正一下：“现在有世界坐标系（世界坐标系也就是物体在A位置的相机坐标系下的3d点位置）”说错了，应为：“现在有世界坐标系（世界坐标系也就是物体在A位兹的相机坐标系）”

同学你好 我还发现一个问题 
Rcw 应该是相机坐标系到世界坐标系的旋转.
也就是Pc=invR(cw)*Pw