gaoxiang12/slambook #163: ch8 编译成功，运行segmentation fault (core dumped)

roachsinai, 2018-12-04:
```
➜  ch8 git:(master) ✗ ./directMethod/cmake-build-debug/direct_sparse data 
*********** loop 0 ************
*********** loop 1 ************
edges in graph: 1402
iteration= 0     chi2= 6293572.221433    time= 0.000394696       cumTime= 0.000394696    edges= 1402     schur= 0        lambda= 965806.498992   levenbergIter= 1
iteration= 1     chi2= 5888155.828834    time= 0.000374733       cumTime= 0.000769429    edges= 1402     schur= 0        lambda= 321935.499664   levenbergIter= 1
iteration= 2     chi2= 5548017.147678    time= 0.000342183       cumTime= 0.00111161     edges= 1402     schur= 0        lambda= 107311.833221   levenbergIter= 1
iteration= 3     chi2= 5290937.995946    time= 0.000349795       cumTime= 0.00146141     edges= 1402     schur= 0        lambda= 35770.611074    levenbergIter= 1
iteration= 4     chi2= 5207371.610703    time= 0.000346302       cumTime= 0.00180771     edges= 1402     schur= 0        lambda= 11923.537025    levenbergIter= 1
iteration= 5     chi2= 5156306.765017    time= 0.00034559        cumTime= 0.0021533      edges= 1402     schur= 0        lambda= 3974.512342     levenbergIter= 1
iteration= 6     chi2= 5133807.422806    time= 0.000415124       cumTime= 0.00256842     edges= 1402     schur= 0        lambda= 1324.837447     levenbergIter= 1
iteration= 7     chi2= 5124650.368695    time= 0.000407728       cumTime= 0.00297615     edges= 1402     schur= 0        lambda= 441.612482      levenbergIter= 1
iteration= 8     chi2= 5112774.456392    time= 0.000379534       cumTime= 0.00335569     edges= 1402     schur= 0        lambda= 147.204161      levenbergIter= 1
iteration= 9     chi2= 5103863.403139    time= 0.000351816       cumTime= 0.0037075      edges= 1402     schur= 0        lambda= 49.068054       levenbergIter= 1
iteration= 10    chi2= 5085463.721599    time= 0.000362203       cumTime= 0.0040697      edges= 1402     schur= 0        lambda= 16.356018       levenbergIter= 1
iteration= 11    chi2= 5052922.896857    time= 0.000361754       cumTime= 0.00443146     edges= 1402     schur= 0        lambda= 5.452006        levenbergIter= 1
iteration= 12    chi2= 4972629.841316    time= 0.000359205       cumTime= 0.00479066     edges= 1402     schur= 0        lambda= 1.817335        levenbergIter= 1
iteration= 13    chi2= 4900169.515522    time= 0.000450973       cumTime= 0.00524164     edges= 1402     schur= 0        lambda= 0.605778        levenbergIter= 1
iteration= 14    chi2= 4888003.377098    time= 0.000404279       cumTime= 0.00564592     edges= 1402     schur= 0        lambda= 0.201926        levenbergIter= 1
iteration= 15    chi2= 4887020.906430    time= 0.000359781       cumTime= 0.0060057      edges= 1402     schur= 0        lambda= 0.134617        levenbergIter= 1
iteration= 16    chi2= 4886717.215284    time= 0.00104467        cumTime= 0.00705037     edges= 1402     schur= 0        lambda= 24090727.650511         levenbergIter= 8
iteration= 17    chi2= 4885996.489698    time= 0.000356024       cumTime= 0.00740639     edges= 1402     schur= 0        lambda= 8030242.550170  levenbergIter= 1
iteration= 18    chi2= 4885632.694219    time= 0.000384733       cumTime= 0.00779113     edges= 1402     schur= 0        lambda= 5353495.033447  levenbergIter= 1
iteration= 19    chi2= 4885625.691907    time= 0.000655758       cumTime= 0.00844688     edges= 1402     schur= 0        lambda= 228415788.093730        levenbergIter= 4
iteration= 20    chi2= 4885387.309209    time= 0.000346492       cumTime= 0.00879338     edges= 1402     schur= 0        lambda= 76138596.031243         levenbergIter= 1
iteration= 21    chi2= 4885212.916240    time= 0.000345564       cumTime= 0.00913894     edges= 1402     schur= 0        lambda= 25379532.010414         levenbergIter= 1
iteration= 22    chi2= 4885046.727489    time= 0.000345444       cumTime= 0.00948438     edges= 1402     schur= 0        lambda= 8459844.003471  levenbergIter= 1
iteration= 23    chi2= 4884831.772796    time= 0.000345987       cumTime= 0.00983037     edges= 1402     schur= 0        lambda= 2819948.001157  levenbergIter= 1
iteration= 24    chi2= 4884332.947280    time= 0.000356112       cumTime= 0.0101865      edges= 1402     schur= 0        lambda= 939982.667052   levenbergIter= 1
iteration= 25    chi2= 4883825.913749    time= 0.000345733       cumTime= 0.0105322      edges= 1402     schur= 0        lambda= 313327.555684   levenbergIter= 1
iteration= 26    chi2= 4883383.826737    time= 0.000344057       cumTime= 0.0108763      edges= 1402     schur= 0        lambda= 104442.518561   levenbergIter= 1
iteration= 27    chi2= 4883030.529985    time= 0.000345956       cumTime= 0.0112222      edges= 1402     schur= 0        lambda= 34814.172854    levenbergIter= 1
iteration= 28    chi2= 4882854.735927    time= 0.000345746       cumTime= 0.011568       edges= 1402     schur= 0        lambda= 11604.724285    levenbergIter= 1
iteration= 29    chi2= 4882587.766097    time= 0.000346863       cumTime= 0.0119148      edges= 1402     schur= 0        lambda= 3868.241428     levenbergIter= 1
[1]    22954 segmentation fault (core dumped)  ./directMethod/cmake-build-debug/direct_sparse data
```

`slambook/ch8/directMethod/direct_sparse.cpp:261:26: 警告：在有返回值的函数中，控制流程到达函数尾 [-Wreturn-type]`

不知有没有遇到同样问题的小伙伴们。。

roachsinai, 2018-12-04:
https://github.com/RainerKuemmerle/g2o/issues/282

starfish55555, 2020-01-08:
请问你解决了没有，同样的问题

dingcong1993, 2021-02-24:
在poseEstimationDirect那个函数最后加一个返回值return true可以解决