gaoxiang12/slambook #166: g2o_viewer not found on Mac

RunJianChen, 2018-12-07:
I install g2o in the 3rdParty on my mac but cannot find g2o_viewer.

Is there any solution to this problem?

Thanks a lot!

gaoxiang12, 2018-12-07:
I have not tested it on mac. g2o_viewer needs libqlgviewer, you may search if grew supports it or compile it from source.

RunJianChen, 2018-12-07:
Thanks a lot!

    I installed libqlgviewer previously and reinstalled g2o, but it didn't work.

    I will try more to fix it but maybe it is a problem on macOS,

    It's really nice of you and thank you again!

IvyGongoogle, 2019-08-26:
@RunJianChen  meet the same problem. How you solve it?

RunJianChen, 2019-08-27:
@IvyGongoogle I thought there was no g2o_viewer for mac and gave up. But other parts worked well. If you need g2o_viewer, maybe you should turn to ubuntu. Thank you!

closed 2019-08-27 by RunJianChen