gaoxiang12/slambook #170: 请问如何判断T是Tcw还是Twc？

Zkjoker, 2018-12-15:
目前学到第五章，101页，第69行。算T*point，即由相机坐标系变为世界坐标系，那这里的T即是Twc。可是书第三讲61页习题第七题，根据q和t恢复出的T是Tcw，请问到底什么情况下求出的是Tcw，什么时候是Twc呢？初学者的一点小疑惑，谢谢高博和各位大神的解答！

xmyqsh, 2018-12-15:
T*point, 是把相机坐标系下的坐标点(point)，变换到世界坐标系下，即Twc代表把相机坐标系下的点变换为世界坐标系下的点的变换。
Tcw，如第七题描述，是世界坐标系(的点)到相机坐标系(的点)的变换。

令pw，pc代表同一个点分别在世界坐标系和相机坐标系下的坐标；
若pw=T*pc，则T就是Twc；
若pc=T*pw，则T就是Tcw；
Twc和Tcw正是互逆的，也是互为转置的，因为是单位正交阵。
根据上面两个式子，就可以确定T是Twc还是Tcw了。
但是，如果函数是封装好的动态链接库，看不到源码，只要求传一个T，这时就没法判断了，正确的方法是Twc和Tcw都试一下。

另外，注意一点，坐标系中数据点的坐标变换和坐标系本身的变换也是互逆的。

Zkjoker, 2018-12-19:
谢谢！！

closed 2018-12-24 by Zkjoker

memememeda, 2019-03-02:
> 谢谢！！

老哥，这里的确有问题，得看实际情况。

假设坐标系c1和坐标系c2， 且c1坐标系 绕z轴旋转90° （右手定则旋转，右手坐标系）到c2坐标系。
那么c1坐标系下的点Pc1（1，0，0） 即 此时x轴方向的方向向量 ，在c2坐标系下的坐标是Pc2（0，-1，0）。
matlab 中 ，输入 inv(rotz(90))* Pc1'   ，才会等于Pc2。
也就是 Rc2c1（即rotz(90)，从c1坐标系到c2坐标系的旋转）的逆 乘以Pc1 = Pc2，
也就是   Rc2c1*Pc2=Pc1

memememeda, 2019-03-14:
> 谢谢！！
https://blog.csdn.net/lixujie666/article/details/82153503  一、1.  有讲 ：
T表示A到B的变换
则Pa=T*Pb