gaoxiang12/slambook #172: ch7运行3d2d文件时报错

LiJianZZZ, 2018-12-18:
高博您好，我在运行pose_estimation_3d2d文件时，出现以下错误：
lijianzhongzz@ubuntu:~/Downloads/slambook-master/ch7$ build/pose_estimation_3d2d 1.png 2.png d1.png d2.png
-- Max dist : 95.000000 
-- Min dist : 4.000000 
一共找到了79组匹配点
段错误 (核心已转储)
（不好意思，我是一名学车辆的大四学生，未来研究生导师让我学习这本书的，但是基础太差了，希望不要嫌弃我）

gaoxiang12, 2018-12-18:
你应该首先用debug模式放个断点看下停在哪里，段错误一般是由于内存错误导致的。

LiJianZZZ, 2018-12-19:
他最后停在了这一行，请问高博可以给我一点指点吗？ 
  ushort d = d1.ptr<unsigned short> (int ( keypoints_1[m.queryIdx].pt.y )) [ int ( keypoints_1[m.queryIdx].pt.x ) ];
应用程序启动器显示：
*** 程序收到了信号 SIGSEGV(Segmentation fault) ***

LiJianZZZ, 2018-12-19:
对不起，高老师。我发现我把变量名写错了

lostwindsenril, 2021-11-29:
> 对不起，高老师。我发现我把变量名写错了

我也遇到你这个问题，但是没理解你说的“变量名写错”是啥意思
后来才发现书上的深度图文件名是d1.img，实际的文件名是1_depth.png。你应该说“传入参数写错了”