gaoxiang12/slambook #177: 关于slam中尺度问题的信息请教·

fangxu622, 2019-01-02:
高老师，您好··我想向您请教一个slam 位姿信息尺度问题，我目前采用 tango手机上 rtab-map app采集位姿信息数据，保存为数据，通过rtab软件打开·导出位姿信息 如图

[image]
[image]

所以我想知道这个位姿薪资如何映射到真实的世界尺度，还是他就是真实世界的尺度，单位是米 或者 厘米··无论是米还是厘米，根据与真实世界的尺度不匹配··？如何把这个数据映射到真实的世界尺度呢··

fangxu622, 2019-01-02:
@gaoxiang12 @tyuownu  hi·~多谢指教哦·~

这是使用脚本画出来的轨迹 只画了x y 

[image]

这是rtabmap 里面显示的轨迹

[image]

gaoxiang12, 2019-01-02:
RTAB是RGBD的，这个就是真实尺度。

kidycharon, 2019-01-09:
@gaoxiang12  @fangxu622高老师您好： 请问，单目ORB_SLAM2 的 KeyFrameTrajectory.txt 文件是怎么对应到真实世界的呢，它的误差精度能达到厘米级么？