gaoxiang12/slambook #181: ch7 make 编译报错，

punnpkin, 2019-01-22:
~/slambook/ch7/build$ make
Scanning dependencies of target pose_estimation_3d3d
[  8%] Building CXX object CMakeFiles/pose_estimation_3d3d.dir/pose_estimation_3d3d.cpp.o
/home/yangnan/slambook/ch7/pose_estimation_3d3d.cpp: In function ‘void find_feature_matches(const cv::Mat&, const cv::Mat&, std::vector<cv::KeyPoint>&, std::vector<cv::KeyPoint>&, std::vector<cv::DMatch>&)’:
/home/yangnan/slambook/ch7/pose_estimation_3d3d.cpp:161:49: error: no matching function for call to ‘cv::ORB::create(’
     Ptr<FeatureDetector> detector = ORB::create();
                                                 ^
In file included from /home/yangnan/slambook/ch7/pose_estimation_3d3d.cpp:3:0:
/usr/local/include/opencv2/features2d/features2d.hpp:269:35: note: candidate: static cv::Ptr<cv::Feature2D> cv::Feature2D::create(const string&)
     CV_WRAP static Ptr<Feature2D> create( const string& name );
                                   ^
/usr/local/include/opencv2/features2d/features2d.hpp:269:35: note:   candidate expects 1 argument, 0 provided
/home/yangnan/slambook/ch7/pose_estimation_3d3d.cpp:162:55: error: no matching function for call to ‘cv::ORB::create(’
     Ptr<DescriptorExtractor> descriptor = ORB::create();
                                                       ^
In file included from /home/yangnan/slambook/ch7/pose_estimation_3d3d.cpp:3:0:
/usr/local/include/opencv2/features2d/features2d.hpp:269:35: note: candidate: static cv::Ptr<cv::Feature2D> cv::Feature2D::create(const string&)
     CV_WRAP static Ptr<Feature2D> create( const string& name );
                                   ^
/usr/local/include/opencv2/features2d/features2d.hpp:269:35: note:   candidate expects 1 argument, 0 provided
CMakeFiles/pose_estimation_3d3d.dir/build.make:62: recipe for target 'CMakeFiles/pose_estimation_3d3d.dir/pose_estimation_3d3d.cpp.o' failed
make[2]: *** [CMakeFiles/pose_estimation_3d3d.dir/pose_estimation_3d3d.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/pose_estimation_3d3d.dir/all' failed
make[1]: *** [CMakeFiles/pose_estimation_3d3d.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

已经按照 CMakelists.txt 的要求 # use this if in OpenCV2 更改 但还是报错

punnpkin, 2019-01-22:
求问有遇到相同问题的同行吗

punnpkin, 2019-01-22:
已经解决 打扰了
https://blog.csdn.net/qq_23225073/article/details/78452638

leekaka, 2019-01-25:
你的ch4能编译过么

leekaka, 2019-01-26:
> 已经解决 打扰了
> https://blog.csdn.net/qq_23225073/article/details/78452638
ch4你是怎么编译通过的呀

punnpkin, 2019-03-08:
> > 已经解决 打扰了
> > https://blog.csdn.net/qq_23225073/article/details/78452638
> > ch4你是怎么编译通过的呀

抱歉这么晚才看到  记不太清了 回头看看