gaoxiang12/slambook #183: ch4 为什么能用左扰动模型来求导啊？

353264241, 2019-01-29:
高博你好，75页提到可以对R进行一次扰动△R来求导，为什么这里可以这样做呢?

gaoxiang12, 2019-02-05:
我不太明白这个问题的意思。为什么不能这么做呢？
按照定义，左乘一个扰动，然后令扰动趋于零，求目标函数相对于扰动的变化率，作为导数来使用。
同时，在优化过程中，用这种导数算出来的增量，以左乘形式更新在当前估计上，于是使估计值一直在SO(3)或SE(3)上。这种手段称为“流形上的优化”。

353264241, 2019-02-11:
这里定义是专属李群的求导法则吗？还是所有函数都可以用这种乘法方式求导

gaoxiang12, 2019-02-13:
这可以理解为一种特殊的定义。
事实上我们在优化中希望计算的是扰动大小，所以求目标函数关于这个扰动的导数。在其他问题里面可能没有扰动的说法，所以谈不上这种特殊定义的导数。

opendvd, 2019-04-24:
你好，如果使用高斯牛顿法算增量Hdeltax = g，那这个增量deltax是扰动的增量还是扰动？
【按照定义，左乘一个扰动，然后令扰动趋于零，求目标函数相对于扰动的变化率，作为导数来使用。
同时，在优化过程中，用这种导数算出来的增量，以左乘形式更新在当前估计上，于是使估计值一直在SO(3)或SE(3)上。这种手段称为“流形上的优化”。】

opendvd, 2019-04-24:
> 我不太明白这个问题的意思。为什么不能这么做呢？
> 按照定义，左乘一个扰动，然后令扰动趋于零，求目标函数相对于扰动的变化率，作为导数来使用。
> 同时，在优化过程中，用这种导数算出来的增量，以左乘形式更新在当前估计上，于是使估计值一直在SO(3)或SE(3)上。这种手段称为“流形上的优化”。

你好，如果使用高斯牛顿法算增量Hdeltax = g，那这个增量deltax是扰动的增量还是扰动？
谢谢高博

MaxTianyy, 2019-04-26:
Hdeltax = g， 这个增量deltax是 扰动。

Haodong-Alex-Yao, 2022-08-09:
我猜测题主想问的是，如果我们想要优化状态量R的话，应该是计算d(Rp) / d(fi), 为什么可以用d(Rp) /  d(delta(fi))代替？也就是为什么变成了对扰动的小量求导而不是对状态量本身求导。  @gaoxiang12

BTREE-C802, 2022-08-09:
您好，我叫旷鸿章，我已经收到了你的邮件。谢谢！

luodanping, 2023-11-07:
> 我猜测题主想问的是，如果我们想要优化状态量R的话，应该是计算d(Rp) / d(fi), 为什么可以用d(Rp) / d(delta(fi))代替？也就是为什么变成了对扰动的小量求导而不是对状态量本身求导。 我的理解是，这里做了一个叫“lifting”的操作，也就是将误差项转变成了关于delta(fi)的方程, 所以这时候我们需要对delta(fi)求导以对新的方程进行最小化。 具体这样做的好处，还想请教一下@gaoxiang12

我理解 x的增量 dx，和dx的的增量，记做d(dx)，当dx是0开始增加的，所谓的error state，这两个增量是一样大的

luodanping, 2023-11-07:
您好，      您的邮件我已收到，我将尽快回复您。