gaoxiang12/slambook #185: 第七章   pose_estimation_3d2d   pose_estimation_3d3d  运行有问题

suphill, 2019-02-17:
第七章的六个程序编译没有问题
如：feature_extraction.cpp, triangulation.cpp ,pose_estimation_2d2d.cpp  ,pose_estimation_3d2d.cpp  ,pose_estimation_3d3d.cpp 。
在运行时feature_extraction.cpp, triangulation.cpp ,pose_estimation_2d2d.cpp都可以。但是，运行pose_estimation_3d2d.cpp和pose_estimation_3d3d.cpp 就有问题。
如：pose_estimation_3d2d 1.png 2.png 1_depth.png 2_depth.png 
反馈的结果为：“crashed with return code:0”
如果不加参数如：pose_estimation_3d2d，得到的反馈结果如上。
这些程序，我都没改动。望能解决，谢谢！

suphill, 2019-02-17:
原因：系统里有2个g2o的库，建议卸载，用书中带的库，可以解决问题。
自问自答，见笑了。

fenneishi, 2019-04-14:
我也遇到了类似的问题：
[g20报错](https://blog.csdn.net/weixin_36673043/article/details/85345886)

gaoxiang12, 2019-04-14:
@fenneishi 如果使用新版g2o，可以参考slambook2的写法：https://github.com/gaoxiang12/slambook2/tree/master/ch7

fenneishi, 2019-04-16:
赞，收到。