gaoxiang12/slambook #186: ch9 project 0.2 *** Program received signal SIGSEGV (Segmentation fault) *** not working with #27

ZhongKang97, 2019-02-22:
gdb output：
Starting program: /home/nvidia/DISK/CODEing/SLAM/slambook/project/projectV2/bin/run_vo 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x0000007fb32267b0 in ?? () from /usr/lib/aarch64-linux-gnu/libQtGui.so.4

ZhongKang97, 2019-02-22:
it seems that this error is caused by libQtGui but how can i fix it?

ZhongKang97, 2019-02-22:
computer platform is TX2 with Ubuntu16.4 LTS

ZhongKang97, 2019-02-22:
$ man 7 signal |grep SIGSEGV
       signals, such as SIGSEGV and SIGFPE, generated as a consequence of exe‐
       SIGSEGV      11       Core    Invalid memory reference

ZhongKang97, 2019-02-23:
绝望，有大侠出手相救的吗

gaoxiang12, 2019-02-25:
我没有在tx2平台上测试过，这可能是个平台特定的问题（关键是我没tx2啊……）